mantidproject/mantid #31249: Indirect ISIS Diagnostics - The plot sliders have gone missing

robertapplin, 2021-04-28:
### Expected behavior
There should be sliders on the first plot of Indirect ISIS Diagnostics to adjust the background range.

### Actual behavior
These sliders are missing.

### Steps to reproduce the behavior
1. Indirect -> Data Reduction -> ISIS Diagnostics
2. Follow the instructions in this example workflow
https://docs.mantidproject.org/nightly/interfaces/indirect/Indirect%20Data%20Reduction.html#isis-diagnostics-example-workflow

The sliders are missing. Needs to be investigated.

### Platforms affected
Found on windows, probably all

SpencerHowells, 2021-04-28:
@robertapplin I'm not seeing sliders for Peak either. Use 2 ranges needs to be True for background.

martyngigg, 2021-05-04:
Fixed by #31274

PR #31274 (2021-05-04): Indirect ISIS Diagnostics - Fix Missing RangeMarkers on plot
**Description of work.**
This PR fixes an issue where the x range markers on the plot of ISIS Diagnostics had gone missing. The markers were hiding at x=0 and were unmovable being the bounds of the `RangeMarker` class hadn't been set properly.

No tests added because this cannot be tested easily. Also the file in question (IndirectTab) has no tests because it is not MVP.

**To test:**
1. Indirect -> Data Reduction -> ISIS Diagnostics
2. Enter run number 26176 and click enter. Wait for it to load.
3. You should see a black range marke [... cut]
closed 2021-05-04 by martyngigg