mantidproject/mantid #31250: Indirect MSD and other Fitting tabs - Crash when trying to move x range sliders

robertapplin, 2021-04-28:
### Expected behavior
The x sliders should be movable on the plot with crashing. The position of the sliders should also reflect the x range seen in the start and end x spin boxes.

### Actual behavior
Trying to move the x sliders causes a crash on MSD Fit (and I assume the other fitting tabs too as this functionality is shared).

### Steps to reproduce the behavior
1. Indirect -> Data Analysis -> MSD Fit tab
2. Load the data below
3. The black sliders on the plot do not reflect the x range in the spin boxes.
3. Try to drag the black sliders on the plot to change the x range. Crash!

### Platforms affected
Found on windows, probably all

SpencerHowells, 2021-04-28:
@robertapplin All 4 are meant to share a few features but IqtFit may not. Stephen did not have time to finish that off.

SpencerHowells, 2021-04-28:
@robertapplin This is new. Recent Pull?

robertapplin, 2021-04-28:
> @robertapplin This is new. Recent Pull?

I found it on the windows nightly

martyngigg, 2021-05-10:
Fixed by #31306

PR #31306 (2021-05-06): Indirect - Fix MSDFit and IqtFit crash when dragging range selectors
**Description of work.**
This PR fixes a crash on MSD Fit and Iqt Fit when attempting to drag the range sliders on a preview plot. It also ensures that the bounds of the RangeMarkers are set correctly so that they can be dragged to the full extent of the displayed data.

**To test:**
1. Indirect -> Data Analysis -> MSD Fit tab
2. Load the data below
3. The black sliders on the plot do not reflect the x range in the spin boxes.
3. Try to drag the black sliders on the plot to change the x range. There should be no crash!
4. The [... cut]