mantidproject/mantid #31253: Not All Plot Settings Loaded From Project Save

Matt-Cumber, 2021-04-28:
### Expected behavior
Loaded plot should be identical as before

### Actual behavior
The plots are not identical

### Steps to reproduce the behavior
In settings change major and minor tick size
Load some data and create a plot
Save the ptoject
Close mantid
Open mantid and load the project you just saved
The plot that is loaded will not have the same settings

### Platforms affected
all

sf1919, 2021-04-29:
Is #31259 related to this?

joseph-torsney, 2021-04-29:
I think they're the same part of the codebase, but the fix I almost have working for this doesn't seem to fix #31259

sf1919, 2021-04-29:
Thats a shame! Thought I'd mention it in case we could sort out two issues with one fix!

joseph-torsney, 2021-04-29:
Good spot and thank you as wasn't aware of it! Maybe there's a more harmonious solution that will fix both

gemmaguest, 2021-05-21:
Fixed by #31284

PR #31284 (2021-05-21): Fix workbench plot settings not being reproduced on project load
Adds support for saving and loading plots with the following plot options introduced by PR #30121:
- Major and Minor tick length, width, and direction
- Tick and label positions (any combination of the left, right, top, bottom sides of the plot)
- Axes spine widths.

Also ensures previous project files are still able to be loaded, as only certain tick positions were supported.

**To test:**
1. Change some of the plot settings in `File > Settings > Plots`, such as tick direction and length
2. Plot a workspace and save it as a proj [... cut]
closed 2021-05-21 by gemmaguest