mantidproject/mantid #31256: Plotting 3D Contour Plot On Sundials Data Causes Exception

Matt-Cumber, 2021-04-28:
### Expected behavior
Produce an error warning or prevents the data from being loaded

### Actual behavior
Produces the error reporter

### Steps to reproduce the behavior
Load a data file from sundials data directory
Try to do a 3D contour plot 

### Platforms affected
all

martyngigg, 2021-04-29:
@Matt-Cumber Could you clarify what the sundials data is and provide a link?

Is there any error message produced in the log that you can paste in to help whoever picks this up?

Matt-Cumber, 2021-04-29:
Error message shown:
```
Traceback (most recent call last):
  File "/Applications/MantidWorkbenchNightly.app/Contents/MacOS/workbench/plugins/workspacewidget.py", line 184, in _do_plot_3D
    plot_function(workspaces)
  File "/Applications/MantidWorkbenchNightly.app/Contents/MacOS/mantid/plots/plotfunctions.py", line 58, in inner_func
    return func(workspaces, *args, **kwargs)
  File "/Applications/MantidWorkbenchNightly.app/Contents/MacOS/mantidqt/plotting/functions.py", line 353, in plot_contour
    ax.contour(ws, levels=DEFAULT_CONTOUR_LEVELS,
  File "/Applications/MantidWorkbenchNightly.app/Contents/MacOS/mantid/plots/mantidaxes.py", line 43, in wrapper
    func_value = func(self, *args, **kwargs)
  File "/Applications/MantidWorkbenchNightly.app/Contents/MacOS/mantid/plots/mantidaxes.py", line 997, in contour
    return self._plot_2d_func('contour', *args, **kwargs)
  File "/Applications/MantidWorkbenchNightly.app/Contents/MacOS/mantid/plots/mantidaxes.py", line 916, in _plot_2d_func
    plotfunctions_func(self, *args, **kwargs),
  File "/Applications/MantidWorkbenchNightly.app/Contents/MacOS/mantid/plots/axesfunctions.py", line 367, in contour
    return axes.contour(x, y, z, *args, **kwargs)
  File "/Applications/MantidWorkbenchNightly.app/Contents/MacOS/mantid/plots/mantidaxes.py", line 43, in wrapper
    func_value = func(self, *args, **kwargs)
  File "/Applications/MantidWorkbenchNightly.app/Contents/MacOS/mantid/plots/mantidaxes.py", line 997, in contour
    return self._plot_2d_func('contour', *args, **kwargs)
  File "/Applications/MantidWorkbenchNightly.app/Contents/MacOS/mantid/plots/mantidaxes.py", line 923, in _plot_2d_func
    return getattr(Axes, name)(self, *args, **kwargs)
  File "/Applications/MantidWorkbenchNightly.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/matplotlib/__init__.py", line 1599, in inner
    return func(ax, *map(sanitize_sequence, args), **kwargs)
  File "/Applications/MantidWorkbenchNightly.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/matplotlib/axes/_axes.py", line 6422, in contour
    contours = mcontour.QuadContourSet(self, *args, **kwargs)
  File "/Applications/MantidWorkbenchNightly.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/matplotlib/contour.py", line 855, in __init__
    kwargs = self._process_args(*args, **kwargs)
  File "/Applications/MantidWorkbenchNightly.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/matplotlib/contour.py", line 1456, in _process_args
    x, y, z = self._contour_args(args, kwargs)
  File "/Applications/MantidWorkbenchNightly.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/matplotlib/contour.py", line 1514, in _contour_args
    x, y, z = self._check_xyz(args[:3], kwargs)
  File "/Applications/MantidWorkbenchNightly.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/matplotlib/contour.py", line 1551, in _check_xyz
    raise TypeError("Input z must be at least a 2x2 array.")
TypeError: Input z must be at least a 2x2 array.

Continue working.

```

I believe sundials is a directory used for elemental analysis. 

[Sundials.zip](https://github.com/mantidproject/mantid/files/6398020/Sundials.zip)

AnthonyLim23, 2021-05-05:
ah, these data sets are just, `x, y, e` so you cant do a contour plot of the data

AnthonyLim23, 2021-05-05:
I deleted the zip because it is not public data

martyngigg, 2021-05-05:
I don't think this is important enough to do for 6.1?

martyngigg, 2021-05-05:
It checks correctly anyway I guess it just needs a cleaner error capture.

martyngigg, 2021-05-10:
Fixed by #31373

PR #31373 (2021-05-10): Catch Error When Plotting 3D Contour On Data Of Incorrect Size
**Description of work.**
Added try catch when doing a contour plot. Send me a message for the data file to be used for testing.

**To test:**
Load the file
Try to do a contour plot
Previously error reporter would show, now a warning message should be seen

Fixes #31256 

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code conform to the [coding standards](http:// [... cut]
closed 2021-05-10 by martyngigg