mantidproject/mantid #31259: Scientific Formatting on Plotting not saving

sf1919, 2021-04-28:
### Expected behavior
When changing the Tick format on the axis of a plot to Scientific Format this should still be set as this format when project is saved and reopened or run from a script generated from that changed plot.

### Actual behavior
Unless the Log checkbox is ticked then the plot axis is defaulting to Decimal format when doing one of the following:-
- saving the project and re-opening it
- copying script to clipboard and then running it in script editor
- saving script to file and then loading script from file

### Steps to reproduce the behavior

- Load a file
- plot one or more of the spectrum from it. 
- Double click on either the x or y axis to edit the axis. 
- In the screen that appears change Tick Format to Scientific Format. Plot axis will then change to Scientific Format. 
- Then do one of the following:-

- saving the project and re-open it
- copy script to clipboard and then running it in script editor
- save script to file and then load & run script from file

### Platforms affected
Windows 10

martyngigg, 2021-04-28:
@thomashampson As you've been close to this code recently, would you be able to see how big a fix this would be?

thomashampson, 2021-05-06:
I've split this into two issues: this one and #31374

thomashampson, 2021-05-06:
The following matplotlib command is used to set scientific tick format without setting a log scale:

```
from matplotlib.ticker import LogFormatterSciNotation

axes.xaxis.set_major_formatter(LogFormatterSciNotation())
```

PR #31407 (2021-05-14): Restore axis tick format on project save and load
**Description of work.**

Created a dictionary of supported tick formats and applied the correct format to plot axes ticks when the project is loaded.

If an axis tick format is not available in the above dictionary, 'ScalarFormatter' is set as the default format.

**To test:**

-> Load a file
-> Plot one or more of the spectra from it
-> Change an axis tick format by clicking on the axis.  As an example, change an axis tick format to 'scientific format'.
-> Save the project and re-open it

The tick format should be the same (scientific) format [... cut]
closed 2021-05-14 by SilkeSchomann