mantidproject/mantid #31261: Improve error handling message for file not found

sf1919, 2021-04-28:
As noted in Manual Testing #31111 

### Expected behavior
When running python scripts in the Script Editor, if file cannot be found because the path doesn't exist or the filename doesn't exist the error message should be clear enough that the error is because the file cannot be found.

### Actual behavior
The error message that appears is 
`File "<string>", line 7, in <module> File "C:\MantidNightlyInstall\bin\mantid\simpleapi.py", line 141, in Load msg = 'Problem setting "Filename" in {}-v{}: {}'.format(name, algm.name(), algm.version(), -`.

### Steps to reproduce the behavior
1) Clear the directories in Manage User Directories and add a path that doesn't exist on your machine. Then try the following script in the Script editor
`ws1 = Load('MUSR15189')`

or

2) Trying loading a file you know doesn't exist on your system e.g. x and then run the following in the script editor
`ws1 = Load('x')`

### Platforms affected
Windows 10

martyngigg, 2021-04-28:
Thanks. This looks to have crept in this release.

martyngigg, 2021-04-30:
Fixed by #31280

PR #31280 (2021-04-30): Fix error message for simpleapi.Load with invalid property value
**Description of work.**

An undefined variable was used and no unit test was present to catch it. A unit test has been added.

**To test:**

* See the instructions in the linked issue.
* Review the new unit test

Fixes #31261 

*This does not require release notes* because **as it was a regression in this release**

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does t [... cut]
closed 2021-04-30 by martyngigg