mantidproject/mantid #31264: Project Save no longer saves interfaces

Pasarus, 2021-04-28:
There is an issue, that causes project save to no longer recognize that interfaces can be saved using specific encoders.

To recreate:
- Open Workbench
- Create a workspace e.g. `ws = CreateSampleWorkspace()`
- Right click on the workspace and click Show Data
- Save your project using File -> Save Project
- Then attempt to load it, it won't successfully load.

I think this has to do with a change in PR #30129 that causes these windows to not be "top level widgets" and appear when we call `QApplication.topLevelWidgets()` which at present is relied on. Moving to `QApplication.allWidgets()` is a no go, because it will take far too long to run through all of them for each encoder, the run time of project recovery would increase by number of widgets multiplied by the number of encoders present in workbench.

DanielMurphy22, 2021-04-28:
#31124 
It seems that the "Show Data" interface doesn't project save. 

ISIS Reflectometry does work with Project Save.
The Engineering Diffraction interface doesn't reopen, but it does seem to save the tag in the .mtdproj file, so this is maybe a slightly different issue to the "Show Data" interface.

Pasarus, 2021-04-29:
> #31124
> It seems that the "Show Data" interface doesn't project save.
>
> ISIS Reflectometry does work with Project Save.
> The Engineering Diffraction interface doesn't reopen, but it does seem to save [...]

Spoke to @clayton-tom when you save an empty Eng diffraction GUI a required part of the data is not saved, so it fails to re-open I have handed this to him to handle.

martyngigg, 2021-05-04:
Fixed by #31286

PR #31286 (2021-04-30): Fix Project Save not saving python interfaces
**Description of work.**
Some work that was done on the EncoderFactory caused most encoders to be removed during registration, this changes the behavior to ensure uniqueness is still preserved and all encoders are present.

**To test:**
- Load a workspace
- Right click on the workspace and select "Show Data"
- Do this with both a Matrix Workspace and a Table Workspace
- So you have 2 windows open.
- Open an Instrument View window
- Now save out a project, File -> Save Project
- Close workbench
- Open Workbench
- Load the project all 3 windows, Both [... cut]
closed 2021-05-04 by martyngigg