mantidproject/mantid #31266: Sliceviewer spectrum axis does not zoom or pan correctly.

thomashampson, 2021-04-28:
### Expected behavior
The 2D plot data should pan and zoom consistently with the axis limits, i.e. the data on the plot should be in the correct position relative to the axes.

### Actual behavior
When panning, the plot data is stretched and compressed in the spectrum axis direction. When zooming, the plot data does not zoom consistently with the spectrum axis range. The time-of-flight axis appears to be fine.

### Steps to reproduce the behavior
- Load a file, e.g. MAR11060.raw from the training course data.
- Open the slice viewer.
- Select the "pan axes" tool.
- Drag the plot upwards.
You will see that the data is stretched and that data is left in negative spectrum space. Hovering the cursor over these negative spectrum values will throw runtime errors, as mentioned in #30677 (related issue).

### Platforms affected
Seen on both windows and linux in Mantid Workbench v6.0 and above.

RichardWaiteSTFC, 2021-04-28:
I can reproduce this, if you keep on going with the pan you eventually get an empty plot and this error
```
Traceback (most recent call last):
  File "C:\mantid\external\src\ThirdParty\lib\python3.8\lib\site-packages\matplotlib\cbook\__init__.py", line 216, in process
    func(*args, **kwargs)
  File "C:\mantid\qt\python\mantidqt\widgets\sliceviewer\cursor.py", line 65, in on_mouse_move
    self.on_cursor_at(xdata, ydata)
  File "C:\mantid\qt\python\mantidqt\widgets\sliceviewer\imageinfowidget.py", line 66, in _on_cursor_at_axesimage
    self._widget.cursorAt(xdata, ydata, arr[i, j])
RuntimeError: indexOfValue - value out of range
```

thomashampson, 2021-04-28:
I've edited the issue, since it applies to colorfill plots in general and not just the slice viewer.

RichardWaiteSTFC, 2021-04-28:
If it affects both colorfills and sliceviewer it might be a bug in here
`C:\mantid\Framework\PythonInterface\mantid\plots\resampling_image\samplingimage.py`

PR #31289 (2021-04-30): Fix problem with spectrum axis when panning on colour fill plots
**Description of work.**

When panning along the spectrum axis on a colour fill plot the data was being stretched instead of panning. If you now pan beyond the spectrum axis limits (eg to negative spectrum values) you see white\blank which is the same as happens if you pan beyond the x axis limits

**To test:**

See test on the linked issue

Fixes #31266, fixes #30677

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review ## [... cut]
PR #31308 (2021-04-30): Fix problem with spectrum axis when panning on colour fill plots
**Description of work.**

Copy of PR #31289 but this time against release-next. See original PR for test script and approval

**To test:**

See original PR

Fixes #31266.

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code conform to the [coding standards](http://developer.mantidproject.org/Standards/)?
- Are the unit tests small and test the class in isolatio [... cut]