mantidproject/mantid #31285: Slicepoint of sliceviewer pinned to bin centers not re-calling BinMD

RichardWaiteSTFC, 2021-04-29:
### Expected behavior
When entering a value for the center of the integrated dimension in sliceviewer it would call BinMD (for MDHisto workspace)
[image]

### Actual behavior
Presently it just plots the slice at the nearest  bin center along that dimension

martyngigg, 2021-05-05:
Moving to next release as it's a change in behaviour that we don't want to risk now that testing has been done.

RichardWaiteSTFC, 2022-01-05:
@martyngigg - I would like to re-open this and take a look for 6.3, but perhaps there is something more urgent?

martyngigg, 2022-01-05:
@RichardWaiteSTFC I'm happy with that. There isn't anything more urgent yet.

martyngigg, 2022-01-10:
Fixed by #33203

closed 2022-01-10 by martyngigg