mantidproject/mantid #31294: Validate calibration/focus file inputs on entry

clayton-tom, 2021-04-29:
From @Pasarus https://github.com/mantidproject/mantid/issues/31116

For the file inputs on both tabs, the runs/files aren't validated until the next user action after entering, which is normally clicking the calibrate or focus button. It would be preferable to validate these as soon as they are entered to avoid the warning box appearing when the user clicks the button.

martyngigg, 2021-08-24:
I've rolled this over to S1 of the next release so it can be assessed as part of the regular sprint planning.

DannyHindson, 2021-10-21:
@RichardWaiteSTFC - just doing a bit of prep for our sprint planning tomorrow. For me the run number fields on the Calibration and Focus tabs are validated as soon as you tab out of the field. This can sometimes take a couple of seconds so if you're v quick off the mark and press the Calibrate or Focus button you do sometimes get a message saying "still validating" or similar. But seems like this is already implemented in the best way we can achieve. Do you agree? If yes I'll close this