mantidproject/mantid #31303: Add Systemtest to check floating interfaces launch

DanielMurphy22, 2021-04-30:
As noted in #31265 there should be a system test to check that all interfaces open when the Additional Window Behaviour is set to `Floating`. 

Currently the test only checks when it is set to `On Top` as this is the default. This issue is to add this test.

DanielMurphy22, 2021-11-03:
I do think this is worth adding!

robertapplin, 2022-05-10:
This is still worth adding

sf1919, 2024-01-23:
Not stale

jhaigh0, 2024-06-12:
@robertapplin do you know if when this issue says "Currently the test..." it is referring to `CppInterfacesStartupTest.py`?

robertapplin, 2024-06-13:
> @robertapplin do you know if when this issue says "Currently the test..." it is referring to `CppInterfacesStartupTest.py`?

Yes I think its referring to both `CppInterfacesStartupTest` and `PythonInterfacesStartupTest`

PR #37530 (2024-07-04): Run open interface system tests with both window behaviours
### Description of work

Test interface opening with both kinds of additional window behaviour.

Fixes #31303

### To test:

- Code review
- Tests should pass

---

### Reviewer

Please comment on the points listed below ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)).
**Your comments will be used as part of the gatekeeper process, so please comment clearly on what you have checked during your review.** If changes are made to the PR during the review process then your final comment will be the most  [... cut]
closed 2024-07-04 by SilkeSchomann