mantidproject/mantid #31327: Add empty background removal possibility to ISIS direct inelastic reduction

abuts, 2021-05-03:
**Original reporter:** [STFC]/[<ross.stewart@stfc.ac.uk>]

New instruments have guides, which produce specific constant background during experiment.  The background is not affected by sample and can be measured on empty instrument

If this background is indeed measured, it can/should be removed from the results of inelastic data reduction

AnthonyLim23, 2021-05-04:
@SpencerHowells I thought we already took this sort of thing into account or am I mistaken?

SpencerHowells, 2021-05-04:
@AnthonyLim23 Our Indirect QENS instruments have no empty background (or didn’t when we began) so we don’t need to do it. Our Reduction has it as an option & the plan is to remove it because it is never used. I don’t know about TOSCA.

abuts, 2021-05-04:
@AnthonyLim23; @SpencerHowells  -- this is entirely for direct inelastic. This is the background removal from empty instrument caused by guides and we have just measured it. Do not think its relevant to anything else.

PR #31408 (2021-05-13): inelastic background for isis instruments
**Description of work.**

There is bug or rather one missing flowpath in the previous PR [#31396](https://github.com/mantidproject/mantid/pull/31396). The sample workspace may indeed come normalized. There is an optimization within the diag I've forgotten about. 

**To test:**

Fixes #31327. 

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code conform to the [coding standards](http: [... cut]
closed 2021-05-13 by martyngigg