mantidproject/mantid #31335: Bug in Save/Load ISaw detector calibration

RichardWaiteSTFC, 2021-05-04:
### Expected behavior
If you Save the detector calibration and load the same calibration the workspace should be unchanged.

### Actual behavior
It appears to invert to flip the detector? Note this ISaw format originates from legacy SNS format and the algorithm `LoadIsawDetCal` has  some WISH specific code to make it work for WISH, maybe these are required for SXD.
[image]

### Steps to reproduce the behavior

(1) Run this script
```
Load(Filename='SXD23767.raw', OutputWorkspace='SXD23767', LoaderName='LoadRaw', LoaderVersion='3')
CloneWorkspace(InputWorkspace='SXD23767', OutputWorkspace='SXD23767_to_detcal')
SaveIsawDetCal(InputWorkspace='SXD23767_to_detcal', Filename='SXD23767_detcal')
LoadIsawDetCal(InputWorkspace='SXD23767_to_detcal', Filename='SXD23767_detcal')
```
(2) View both workspaces in the Instrument Viewer (screenshot above is bank 4)

### Platforms affected

DannyHindson, 2021-11-23:
@peterfpeterson - are you interested in getting this problem fixed? We don't use this algorithm at ISIS but noticed it while running it on an ad hoc basis while investigating a problem. We don't have capacity to fix it though so will perhaps close it if nobody's interested

peterfpeterson, 2021-11-23:
We are looking into it to see if it is an issue

closed 2022-03-08 by DannyHindson