mantidproject/mantid #31340: Update all external links in codebase

martyngigg, 2021-05-04:
The code contains many external links to www.mantidproject.org. We aim to shrink the size of `www` domain. All user docs should go to `docs` folder and developer docs to `dev-docs` folder in the code. This will also allow the offline help to be more useful.

This task is to assess all of the `www.mantidproject.org` links in the codebase and find new homes for most of them.

Matt-Cumber, 2021-05-24:
Links which lead to 404 page not found error, keeping this updated:

Link :http://www.mantidproject.org/CreateBackToBackParametersGSAS
File: BackToBackExponetional.rst Line: 36

Link:  http://www.mantidproject.org/Nearest_Neighbours
File: SpatialGrouping-v1.rst Line: 23 

Link: http://www.mantidproject.org/MantidPlot:_SliceViewer
File: BinMD-v1.rst Line 127, 150
File: MDWorkspace Line 102

Link: http://www.mantidproject.org/SpecialWorkspace2D
File: MaskDetectors-v1.rst Line 118, 122
File: SaveMask-v1.rst Line 36, 39

Link: http://www.mantidproject.org/MD_Transformation_factory
File: ConvertToMD-v1.rst Line 15
File: ConvertToMDMinMaxGlobal-v1.rst Line 46, 47

Matt-Cumber, 2021-06-11:
Un-assigning myself as started the new sprint and will most likely not get to work on this again

martyngigg, 2022-09-06:
Available as Conor is currently on rotation to another group

PR #35049 (2023-01-30): Update all external links across codebase
**Description of work.**

This work assessed all links pointing to www.mantidproject.org. Those pointing to wiki pages that should have their documentation inside the code base have been moved to rst pages. Others have their addresses updated to point to archive.mantidproject.org

The MantidHelpWindow code has also been simplified so that a page can be called with a relative page path and it is assumed to be relative to the base paths of either the Qt Help or HTML documentation

**To test:**

To check the help window:

* Build the `docs-qthelp` target
 [... cut]
closed 2023-01-30 by DannyHindson