mantidproject/mantid #31349: Model Analysis tab for Muon Analysis

robertapplin, 2021-05-05:
### Description
Muon Scientists would like to be able to perform a fit of:
fit parameter vs fit parameter
sample log vs fit parameter
sample log vs sample log

within the Muon Analysis interface. These fit parameters will be stored in a results table generated by the results tab after performing a fit. 

To create this Model Analysis tab the first few steps are clear:
1. Move the cyclic dataset selector combobox to the BasicFittingWidget (from the GeneralFittingWidget)
2. Create a new Model Analysis tab by creating a ModelAnalysisWidget that inherits from the BasicFittingWidget
3. Add two comboboxes to the ModelAnalysisWidget for selecting X and Y fit parameters or sample logs.
4. Might need to create a context for holding the resulting fit data?

AnthonyLim23, 2021-05-05:
you will also need to add a pane for the plotting of the data

AnthonyLim23, 2021-05-05:
also yes to 4.

robertapplin, 2021-05-05:
Ok, like an external plot pane?

AnthonyLim23, 2021-05-05:
yep. So at the moment we have "data" and "fit". We will add a new one (we can call it "Model" or "Results" or something)

closed 2021-07-20 by AnthonyLim23