mantidproject/mantid #31354: Workbench Hang on start after updating facilties.xml

DavidFair, 2021-05-05:
### Expected behavior
Workbench should gracefully handle updating facilities files 

### Actual behavior
Complete hang, followed by crash on a fresh install

```
 File "C:\MantidNightlyInstall\bin\workbench\app\start.py", line 207, in start
   exit_value = start_workbench(app, options)
 File "C:\MantidNightlyInstall\bin\workbench\app\start.py", line 178, in start_workbench
   AboutPresenter(main_window).show()
 File "C:\MantidNightlyInstall\bin\workbench\widgets\about\presenter.py", line 40, in __init__
   self.setup_facilities_group()
 File "C:\MantidNightlyInstall\bin\workbench\widgets\about\presenter.py", line 103, in setup_facilities_group
   self.action_facility_changed(default_facility)
 File "C:\MantidNightlyInstall\bin\workbench\widgets\about\presenter.py", line 111, in action_facility_changed
   current_value = ConfigService.getFacility().name()
RuntimeError: Facilities search object  
```

### Steps to reproduce the behavior
(This is speculation as I haven't had a chance to reproduce)
- Ensure that facility and default instrument are blank (like on a fresh install)
- Either delete facilties.xml or transplant it from an older version of the instrument definitions
- Start workbench, in log it should state it's updating n files
- Click around, it froze for me immediately on startup

### Platforms affected
Windows, possibly all?

DavidFair, 2021-05-05:
Marking as high priority, as having Workbench hang on first startup after someone has held off upgrading for 3-6+ months doesn't look great

martyngigg, 2021-05-05:
Is this essentially the same bug as #30622?

PR #31364 (2021-05-11): Fix workbench hang on clean install
**Description of work.**
This PR fixes a crash by workbench when starting up a clean install of Mantid workbench with an empty facility.

**To test:**
Follow the testing instructions in the issue and make sure mantid opens ok #31354 

Fixes #31354

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code conform to the [coding standards](http://developer.mantidproject.org/Standards/)?
- Are the unit [... cut]
closed 2021-05-11 by martyngigg