mantidproject/mantid #31358: Deleting peak in peak table workspace causes sliceviewer to delete peak in different table in slicveviewer as well

RichardWaiteSTFC, 2021-05-05:
### Expected behavior
When plotting two peak tables (e.g. one integrate one not), if you delete a peak in one workspace (delete row in peaks workspace of table in ADS) then the table in sliceviewer will be updated and will no longer have that peak, the other table will not be effected.

### Actual behavior
Despite the peak still existing in one of the tables, sliceviewer doesn't display it (i.e. it is removed from both tables in the sliceviewer).

### Steps to reproduce the behavior
(1) Run this script to get some data and peaks
```
ws = CreateMDWorkspace(Dimensions='3', Extents='-5,5,-4,4,-3,3',
                       Names='H,K,L', Units='r.l.u.,r.l.u.,r.l.u.',
                       Frames='HKL,HKL,HKL',
                       SplitInto='2', SplitThreshold='50')
expt_info = CreateSampleWorkspace()
ws.addExperimentInfo(expt_info)
# FakeMDEventData(ws, PeakParams='1e+06,0,1,0,0.5', RandomSeed='3873875')
FakeMDEventData(ws, EllipsoidParams='1e+05,0,1,0,1,0,0,0,1,0,0,0,1,0.01,0.04,0.02,1', RandomSeed='3873875')
p = CreatePeaksWorkspace(InstrumentWorkspace='ws', NumberOfPeaks=0, OutputWorkspace='peaks')
pk = p.createPeakHKL([0,1,0]) # axes aligned with viewing axes
p.addPeak(pk)
IntegratePeaksMD(InputWorkspace='ws', PeakRadius=0.6, BackgroundInnerRadius=0,
    BackgroundOuterRadius=0, PeaksWorkspace='peaks', OutputWorkspace='peaks_int', 
    Ellipsoid=False, UseOnePercentBackgroundCorrection=False)
```
(2) In the sliceviewer overlay both `peaks `and `peaks_int` tables
(3) Open up one table (e.g. `peaks`) from the ADS, right-click the first row and delete
(4) In sliceviewer the peak will have been removed from both tables

### Platforms affected
Tested on IDAaaS

RichardWaiteSTFC, 2021-05-06:
Update: it looks like this is due to a problem with adding multiple peaks workspace and not related directly to the ADS observer which catches the replacement of the workspace, and therefore probably not related to #31367.
You also run into a problem if you delete the peak in one table before even opening sliceviewer:
```
ws = CreateMDWorkspace(Dimensions='3', Extents='-5,5,-5,5,-5,5',
                       Names='Q_lab_x,Q_lab_y,Q_lab_z', Units='U,U,U',
                       Frames='QLab,QLab,QLab',
                       SplitInto='2', SplitThreshold='50')
expt_info = CreateSampleWorkspace()
ws.addExperimentInfo(expt_info)
# add peak
p = CreatePeaksWorkspace(InstrumentWorkspace='ws', NumberOfPeaks=0, OutputWorkspace='peaks')
pk = p.createPeak([1,1,1]) # axes aligned with viewing axes
p.addPeak(pk)
IntegratePeaksMD(InputWorkspace='ws', PeakRadius=0.6, BackgroundInnerRadius=0.6,
    BackgroundOuterRadius=1, PeaksWorkspace='peaks', OutputWorkspace='peaks_int', 
    Ellipsoid=False, UseOnePercentBackgroundCorrection=False)
# delete peak
DeleteTableRows(TableWorkspace='peaks', Rows='0')
```
And overlay both in table and sliceviewer believes both tables have the peak - if you click the peak it then throws an error. Definitely a problem with how multiple peak tables are added.

RichardWaiteSTFC, 2021-05-07:
Having trouble with finding how this is happening - but it is a regression from v6.0!

martyngigg, 2021-05-10:
Fixed by #31392

PR #31392 (2021-05-10): Stop the peaks viewer overwriting all tables with last loaded
**Description of work.**

When multiple peak tables were overlaid in the sliceviewer, the tables would be overwritten with the last table loaded, this was because the constructor of the table view would use the same model for all views if none was explicitly provided.
This bug was introduced in #30791 - thanks @StephenSmith25 who found this fix.

To-Do
- Will add release notes when PRs #31388 and #31391 are merged

**To test:**

(1) Run this script

```
ws = CreateMDWorkspace(Dimensions='3', Extents='-5,5,-5,5,-5,5',
                [... cut]
closed 2021-05-10 by martyngigg