mantidproject/mantid #31362: Changing overlaid peaks table causes crash in sliceviewer if peak already selected

RichardWaiteSTFC, 2021-05-05:
### Expected behavior
When an overlaid peak table is changed the table would be re-loaded and all the peaks re-plotted  - this doesn't appear to be the case, or at least it's not happening correctly.

### Actual behavior

Having overlaid a peak table  **AND SELECTED/CLICKED ON A PEAK **  (otherwise no error) then re-integrating or renaming that table there is an out-of index crash if you then try to click on a peak in the sliceviewer table (note that post-rename the name at the top of the sliceviewer table is updated).
```
Traceback (most recent call last):
  File "/opt/mantidnightly/lib/mantidqt/widgets/sliceviewer/peaksviewer/view.py", line 171, in _on_row_clicked
    self._row_selected()
  File "/opt/mantidnightly/lib/mantidqt/widgets/sliceviewer/peaksviewer/view.py", line 177, in _row_selected
    self._presenter.notify(self._presenter.Event.PeakSelected)
  File "/opt/mantidnightly/lib/mantidqt/widgets/sliceviewer/peaksviewer/presenter.py", line 90, in notify
    self._peak_selected()
  File "/opt/mantidnightly/lib/mantidqt/widgets/sliceviewer/peaksviewer/presenter.py", line 125, in _peak_selected
    self._view.set_axes_limits(*self.model.viewlimits(selected_index), auto_transform=False)
  File "/opt/mantidnightly/lib/mantidqt/widgets/sliceviewer/peaksviewer/model.py", line 102, in viewlimits
    rep = self._representations[index]
IndexError: list index out of range
```

### Steps to reproduce the behavior

### Platforms affected

RichardWaiteSTFC, 2021-05-06:
Update: it looks like this is due to a problem with adding a peaks workspace. If you comment out `self.append_peaksworkspace(ws_name)` on L278 of `qt\python\mantidqt\widgets\sliceviewer\peaksviewer\presenter.py` - the workspace with the peak deleted is removed from the view and the other table still has the peak selected. When you re-overlay the first table it then removes the peak from both view tables. It doesn't look like it's related directly to the ADS observer which catches the replacement of the workspace, and therefore probably not related to #31367 sorry!

RichardWaiteSTFC, 2021-05-06:
To give MWE for the above run 
```
ws = CreateMDWorkspace(Dimensions='3', Extents='-5,5,-5,5,-5,5',
                       Names='Q_lab_x,Q_lab_y,Q_lab_z', Units='U,U,U',
                       Frames='QLab,QLab,QLab',
                       SplitInto='2', SplitThreshold='50')
expt_info = CreateSampleWorkspace()
ws.addExperimentInfo(expt_info)
# add peak
p = CreatePeaksWorkspace(InstrumentWorkspace='ws', NumberOfPeaks=0, OutputWorkspace='peaks')
pk = p.createPeak([1,1,1]) # axes aligned with viewing axes
p.addPeak(pk)
IntegratePeaksMD(InputWorkspace='ws', PeakRadius=0.6, BackgroundInnerRadius=0.6,
    BackgroundOuterRadius=1, PeaksWorkspace='peaks', OutputWorkspace='peaks_int', 
    Ellipsoid=False, UseOnePercentBackgroundCorrection=False)
# delete peak
DeleteTableRows(TableWorkspace='peaks', Rows='0')
```
And overlay both in table and sliceviewer believes both tables have the peak - if you click the peak it then throws an error

RichardWaiteSTFC, 2021-05-06:
It looks like it updates both tables with whatever the last table loaded includes - this is normally ok as tables have the same peaks (i.e. different integrations) - but not in this case!

gemmaguest, 2021-05-17:
Fixed by #31422

closed 2021-05-17 by gemmaguest