mantidproject/mantid #31367: Crash with sliceviewer open on live data

gemmaguest, 2021-05-05:
### Expected behavior
You should be able to view live data with sliceviewer

### Actual behavior
Mantid can crash if sliceviewer is open and the live data workspace updates.

### Steps to reproduce the behavior
Must be on IDAaaS with real live data available.
- Open the ISIS Reflectometry interface
- Set the instrument to INTER
- In the Live Data section at the bottom left, set the interval to `1s` and click `Start monitor`
- When the `TOF_live` workspace appears, open sliceviewer on it
- Mantid then crashes pretty quickly afterwards

### Platforms affected
Only seen on IDAaaaS so far.

Reproducible in nightly and v6.0.0 so it is not a new problem, but users are trying to use this right now so it is an important fix.

I've only been able to reproduce this so far with starting it from the reflectometry interface so it's possible it's related to that, but I suspect it is more likely to do with the more complex processing that is done affecting the timing. It seems to work fine without sliceviewer open so it does implicate sliceviewer.

gemmaguest, 2021-05-06:
I think I've managed to reproduce this (or possibly a different crash) by opening lots of slice viewer plots on the TOF_live workspace

joseph-torsney, 2021-05-12:
We've made some progress on this, but we're not sure it will make 6.1. 

The problem is when you run `LoadInstrument` the first time, the ADS observer detects the workspace update and eventually asks the [view to redraw the canvas](https://github.com/mantidproject/mantid/blob/a8e960f0a63bb8571144e1a459ed33f571f8b6fa/qt/python/mantidqt/widgets/sliceviewer/view.py#L342). While this is happening, another `LoadInstrument` [modifies the workspace again](https://github.com/mantidproject/mantid/blob/a8e960f0a63bb8571144e1a459ed33f571f8b6fa/Framework/DataHandling/src/LoadInstrument.cpp#L216), but the previous draw is still happening on the old workspace and segfaults when it wants to [access some spectrum info](https://github.com/mantidproject/mantid/blob/a8e960f0a63bb8571144e1a459ed33f571f8b6fa/Framework/PythonInterface/mantid/plots/datafunctions.py#L596).

Also, I wasn't able to make it not crash by calling `refresh_view` directly. Interestingly the crash doesn't occur in the `FigureManagerADSObserver`, maybe because [it isn't always redrawing the canvas](https://github.com/mantidproject/mantid/blob/a8e960f0a63bb8571144e1a459ed33f571f8b6fa/qt/applications/workbench/workbench/plotting/figuremanager.py#L135)?