mantidproject/mantid #31381: Muon Analysis - Results table add the Group/pair in its own column

robertapplin, 2021-05-06:
**Original reporter:** James Lord

### Description
James mentioned it would be useful if they could add the Group/Pair of a fit result to the results table to make it easier to read what the result is for rather than looking in the middle of a workspace name.

### Steps to reproduce the behavior
1. Perform a few simultaneous fits on Muon Analysis across Group/Pairs
2. Output a results table on the Results tab. 
3. They want to be able to tick Group/Pair in the results tab to add a Group/Pair column

### Platforms affected
All

AnthonyLim23, 2021-05-07:
Do we want to just makes this a sample log? It will then automatically appear in the results table tab. We can default it to be selected and do some stuff to make sure its at the top of the table.

robertapplin, 2021-05-11:
> Do we want to just makes this a sample log? It will then automatically appear in the results table tab. We can default it to be selected and do some stuff to make sure its at the top of the table.

Yes that sounds like a good idea to me

AnthonyLim23, 2021-05-18:
I believe we copy the sample logs from the original workspace that we fit to. So we should add the sample log in the following 3 (that define the group/pair):
https://docs.mantidproject.org/nightly/algorithms/MuonGroupingAsymmetry-v1.html 
https://docs.mantidproject.org/nightly/algorithms/MuonGroupingCounts-v1.html 
https://docs.mantidproject.org/nightly/algorithms/MuonPairingAsymmetry-v1.html

this may need us to manually clean up the diffs in the GUI back end

PR #32291 (2021-08-23): Allow norm and group name in results tab of muon GUI
**Description of work.**
This allows the user to select the norm and group as part of the results tab.

**To test:**
Open Muon Analysis
Load MUSR 62260
Do a fit
Go to the results tab
You should be able to select "analysis_asymmetry_norm" and "analysis_group"
Check they look good in the modelling tab

Fixes #31381

This also Fixes #31348, as it gives the only other parameter of interest from the fit

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

# [... cut]
closed 2021-08-23 by gvardany