mantidproject/mantid #31403: Drill 6.2

gvardany, 2021-05-10:
- [ ] More shortcuts: Ctrl+U (for ungrouping), Ctrl+D (for data directories), Ctrl+Z (undo last), Ctrl+M (set row as master)
- [ ] Add another special column "Comment". This should not be used in any processing, but should be saved/loaded with the rundex. Users can freely type anything in it.
- [ ] Investigate making drill dockable to workbench.
- [ ] For grouped samples, it puts the row index in the output name if there is no outputworkspace. This should be the row title (e.g. A5) rather than the plain index, since once a row is in a group, we loose the connection to a row index.
- [ ] Silence the warning `No user file was found in the workspace` from SaveCansas1D/2D. Downgrade to notice or information.
- [ ] Add sub-directories for the exports by format
- [ ] Implement possibility of typed columns. Particularly a checkbox for boolean and combo-box for choices will be very useful.

gvardany, 2022-01-21:
This is now superseded by the wish list composed by the SANS group (some items are carried over).

martyngigg, 2022-05-09:
@gui-co I'm just going over issues for 6.4. Does Gagik's comment mean this should be closed now?

martyngigg, 2022-05-10:
I'll move this to 6.5 for now. Feel free to close it if appropriate.

gui-co, 2022-05-10:
This can be closed. Thanks.