mantidproject/mantid #31486: Script Repo fails setup on MacOS

DanielMurphy22, 2021-05-17:
### Expected behavior
Setting up the script repo in a new folder works seemlessly
### Actual behavior
Cannot connect to the server on MacOS
### Steps to reproduce the behavior
In Workbench, go to File > Script Repo
Click OK, then in the file browser, create a new folder, select it and click Open.
[image]

### Platforms affected
MacOS, Catalina and Big Sur, recent nightly and release 6.0

martyngigg, 2021-05-17:
Adding High Priority as we should not release with it in this state.

I just want to confirm we have checked it works on Linux & Windows?

martyngigg, 2021-05-17:
@DavidFair I think this could be to do with the server...

DanielMurphy22, 2021-05-17:
It works for me on Linux and Windows, I've only seen it broken on MacOS, but it very well could be a server problem. Do you see this problem on MacOS @DavidFair ??

martyngigg, 2021-05-17:
It looks like we have an issue with the Jenkins [job](https://builds.mantidproject.org/job/sync_scripts_download/842/console). It was calling `python` straight but changing that to `python3` shows another issue in the link.

martyngigg, 2021-05-17:
> It works for me on Linux and Windows, I've only seen it broken on MacOS, but it very well could be a server problem. Do you see this problem on MacOS @DavidFair ??

That's odd. Maybe we have two problems...

DavidFair, 2021-05-17:
I'm getting the same error here on Catalina with the latest nightly 
Using wget I think I know why the problem is difficult to pin down, it depends on if you have IPv6 as wget / similar fails back to IPv4

Could @Matt-Cumber check if he has IPv6 [here](https://test-ipv6.com/), and equally anyone who it does work for check that they don't have it to confirm the theory

Matt-Cumber, 2021-05-17:
> Could @Matt-Cumber check if he has IPv6 [here](https://test-ipv6.com/), and equally anyone who it does work for check that they don't have it to confirm the theory

Running that check I fail with `No IPv6 address detected` and `Connections to IPv6-only sites are timing out. Any web site that is IPv6 only, will appear to be down to you.`

DavidFair, 2021-05-17:
And it failed for you too?
Well that possibly blows the IPv6 theory out of the water :thinking:

DavidFair, 2021-05-17:
Ah but the VPN will use your home connection wherever it can, so it will access via IPv6 is my hunch

I've just asked @StephenSmith25 to run the test locally too

DavidFair, 2021-05-17:
Stephen doesn't have IPv6 I'm 90% certain this is the offender as were not doing [happy eyeballs](https://en.wikipedia.org/wiki/Happy_Eyeballs) correctly and anyone on site won't notice it as business adoption of IPv6 is....well slow...

So this is almost certainly a server issue that's existed for a while but only coming to boil as more ISPs are migrating end-users

closed 2021-05-17 by DavidFair