mantidproject/mantid #31492: SliceViewer cuts not working for 3D MDWorkspace

mducle, 2021-05-17:
**Original reporter:** A. Krajewska / ISIS

### Expected behavior

Making a ROI cut in SliceViewer and then pressing `'c'` creates 1D MDWorkspaces with the correct cut data.

### Actual behavior

The workspaces are created but the integration range in the 3 dimension (the dimension not shown in the plot) is incorrect, giving an incorrect cut.

### Steps to reproduce the behavior

* Load this dataset: [test_combined.zip](https://github.com/mantidproject/mantid/files/6493961/test_combined.zip)
* Right-click on it and select `Show Slice Viewer`.
* Move the `[0,0,L]` slider to somewhere in the middle (say around -3 [or type `-3` in the end box]).
* Change the color scale max limit to 5e-8 to see the Bragg peaks.
* Click on the "region of interest" icon (6 to the right of the "Home" icon / 4 to the left of the "save" icon).
* Select a rectangle which contains a few peaks - should see that the cuts along `x` (`[H,0,0]`) and `y` (`[0,K,0]`) appear, along with the text `Press key to send roi/cuts to workspaces: r=roi, c=both cuts, x=X, y=Y. Esc clears region`).
* Press `c`
* Two workspaces should be created: `<workspace_name>_cut_x` and `<workspace_name>_cut_y`.
* These workspace should show the same data as in the SliceViewer window side bars but they show nothing (values are all `NaN`s).

What happens seesm to be that the wrong axes limits for the `z` (`[0,0,L]`) direction is sent - looking at the history of the workspaces, the command used to generate them is:

```
IntegrateMDHistoWorkspace(InputWorkspace='MD_merged_normalised', 
    P1Bin='-5.37932,0,-1.12378', P2Bin='2.82863,5.79823', P3Bin='-15,15', 
    OutputWorkspace='MD_merged_normalised_cut_x')
IntegrateMDHistoWorkspace(InputWorkspace='MD_merged_normalised', 
    P1Bin='-5.37932,-1.12378', P2Bin='2.82863,0,5.79823', P3Bin='-15,15', 
    OutputWorkspace='MD_merged_normalised_cut_y')
```

whereas it should be:

```
IntegrateMDHistoWorkspace(InputWorkspace='MD_merged_normalised', 
    P1Bin='-5.37932,0,-1.12378', P2Bin='2.82863,5.79823', P3Bin='-3.225,-2.925', 
    OutputWorkspace='MD_merged_normalised_cut_x')
IntegrateMDHistoWorkspace(InputWorkspace='MD_merged_normalised', 
    P1Bin='-5.37932,-1.12378', P2Bin='2.82863,0,5.79823', P3Bin='-3.225,-2.925', 
    OutputWorkspace='MD_merged_normalised_cut_y')
```

(e.g. `P3Bin` is incorrectly sent by the `c` command; `-15,15` is the full data range but in many slices there are `NaNs` and this means the outputs are also `NaN`s whereas in the specific slice we looked at the data exists and is not `NaN`.

### Platforms affected

Only checked on Linux, but presumably all.

mducle, 2021-11-22:
The issue is still present in the latest nightly `6.2.20211119.1752` except that now the ROI workspaces are not created but an error message:

```
tuple indices must be integer or slices, not float
```

appears.

RichardWaiteSTFC, 2022-01-05:
@martyngigg - it might be good to address this in 6.3. I believe it only affects MDHisto eworkspaces. e.g.
(1) Create MDEvent
```
ws = CreateMDWorkspace(Dimensions='3', Extents='-3,3,-3,3,-3,3',
                       Names='H,K,L', Units='r.l.u.,r.l.u.,r.l.u.',
                       Frames='HKL,HKL,HKL',
                       SplitInto='2', SplitThreshold='10')
```
(2) Open sliceviewer, draw ROI, press c (no error, cuts are made)
(3) Convert to MDHisto 
```
BinMD(InputWorkspace='ws', AxisAligned=False, 
    BasisVector0='H,r.l.u.,1.0,0.0,0.0', 
    BasisVector1='K,r.l.u.,0.0,1.0,0.0', 
    BasisVector2='L,r.l.u.,0.0,0.0,1.0', 
    OutputExtents='-3,3,-3,3,-3,3', OutputBins='33,33,33', 
    NormalizeBasisVectors=False, OutputWorkspace='ws_histo')
```
(4) Repeat (2) - this time get error
`tuple indices must be integers or slices, not float`

Note this happens even if you don't delete the underlying MDEvent workspace

martyngigg, 2022-01-14:
Fixed by #33288

closed 2022-01-14 by martyngigg