mantidproject/mantid #31501: Mantid nightly crash on running MARI reduction scripts.

mducle, 2021-05-17:
### Expected behavior

MARI reduction scripts run ok, and produces `nxspe` files and reduced workspaces.

### Actual behavior

Hard crash (C++ level `Unhandled Exception`) when the script is run.

### Steps to reproduce the behavior

On IDAaaS, run the attached scripts: 
[mari_reduction_files.zip](https://github.com/mantidproject/mantid/files/6495116/mari_reduction_files.zip)

If run on the commandline, it fails with: https://gist.github.com/mducle/6737d23436fe52757f9d339493f7badf

The key error seems to be:

```
********* UNHANDLED EXCEPTION *********
  what(): SpectrumInfo: No detectors for this workspace index. search object 0
```

### Platforms affected

IDAaaS (CentOS 7), others not checked.

DanielMurphy22, 2021-05-18:
As mentioned on Slack, this issue may be caused by https://github.com/mantidproject/mantid/pull/30163

DannyHindson, 2021-05-18:
I've found the cause of this - it is the PR mentioned above relating to changes to ConvertUnits and some of the Unit classes. The MARI reduction script performs a unit conversion from TOF to energy. Spectrum 0 in the workspace doesn't have any detectors and this used to result in the spectrum being masked but my changes had stopped this happening. I've restored the original behaviour and the script now runs fine for me. Will create a PR later on. The problem only affected the "Energy" unit.

I'm having a look at why non of the system tests failed because of this. I think it's because they all run with the "run_diagnostics" setting set to false whereas this is "True" in the MARI instrument parameter file.

mducle, 2021-05-18:
Thanks @DannyHindson ! Yes, I think there is supposed to be a MARI reduction script/test in the system test and was wondering whether I've modified the one we used (which is hosted in the separate `scriptrepository` repo) too much compared to the systems test one. 

The MARI instrument definition is a bit weird in that spectrum 0, 4 and 28 don't actually physicall exist... it's been like this since forever (since before Mantid) so I didn't want to change it... I guess maybe they thought once upon a time to have additional detectors / monitors..

DannyHindson, 2021-05-19:
@mducle - I've just created a PR for this issue. The existing system tests do go through the same code path as the script you supplied (including the run_diagnostics part - so my earlier comment suggesting this was a difference was wrong). The system test didn't catch the crash because the run numbers are different and the crash only happened inside MedianDetectorTest for certain data. So if the script where you found the problem is a fairly standard test script that you use then yes, worth us updating the Mantid system test to use the same run numbers. I haven't done this as part of the PR so suggest we do this as a follow up task

martyngigg, 2021-05-20:
Fixed by #31517

PR #31517 (2021-05-19): Fix gap in validation on the Energy unit
**Description of work.**

A gap in the validation logic for the Energy unit was causing a crash in one of the MARI scripts. The cause was a spectrum without any detectors that couldn't undergo a unit conversion to Energy due to the absence of an L2 value. This was not being masked in a call to ConvertUnits and a later part of the MARI script wasn't handling this situation properly.

Also tighten up unit tests on the validations applied on various units

GetAllEi wasn't supplying an L2 value when doing an energy unit conversion because it stuffs L_tot in [... cut]
closed 2021-05-20 by martyngigg