mantidproject/mantid #31586: Instrument View: Error unwrapping surface for some instruments

martyngigg, 2021-05-26:
### Expected behavior

Instrument view shows unwrapped view of instrument.

### Actual behavior

An error is generated for some IDFS, e.g CRISP & INTER.

### Steps to reproduce the behavior

* Start Workbench
* Run `LoadEmptyInstrument` and set `InstrumentName` to `CRISP` and set a workspace name
* Click show instrument on the workspace
* Click the box showing "Full 3D" and switch it to one of the other flat projections like Cylinder Y
* An error dialog is generated.

### Platforms affected

All

DannyHindson, 2021-05-26:
I've just tried this out - I see the problem in the latest code but I also see it in v6.0. So doesn't look like a regression to me:

[image]

martyngigg, 2021-05-26:
Oh. I wonder why it seemed to work for me in 6.0.

Would you still mind having a look at how large/complex a fix might be to see if it's safe to include?

DannyHindson, 2021-05-26:
I've had a look into this. I think this can be fixed with a couple of one line changes:

- I think the warning can simply be removed. It's created in a function that tries to spot if a bank of detectors has been split into two when it's unwrapped into a rectangle using a cylindrical projection. For a bank of detectors with width 1 (like in CRISP, INTER) the function doesn't need to do anything
- The values u_min and u_max are currently set to 0 and 1 respectively immediately after the warning. Since u is measured in radians, the value of u_max=1 doesn't make much sense. I suggest setting u_max=0. I've tried this and it works OK

```
void RotationSurface::findAndCorrectUGap() {
  double period = uPeriod();
  if (period == 0.0)
    return;
  const int nbins = 1000;
  std::vector<bool> ubins(nbins);
  double bin_width = fabs(m_u_max - m_u_min) / (nbins - 1);
  if (bin_width == 0.0) {
    QApplication::setOverrideCursor(QCursor(Qt::ArrowCursor));
    QMessageBox::warning(
        nullptr, tr("Mantid - Instrument view warning"),
        tr("Rotation surface: failed to build unwrapped surface"));
    QApplication::restoreOverrideCursor();
    m_u_min = 0.0;
    m_u_max = 1.0;
    return;
  }
```

A few commits that I used:
findAndCorrectUGap added: https://github.com/mantidproject/mantid/commit/dd96095e6c0ee7e965bf26e48702f5556152993f
exception added: https://github.com/mantidproject/mantid/commit/82b17a1c7b704c352a7abf37384818fcc15749e3
downgraded to a warning: https://github.com/mantidproject/mantid/commit/a3b891397fc7c689a31e97e19ece8075549634c2

martyngigg, 2021-05-26:
Thanks for investigating this.

> I've had a look into this. I think this can be fixed with a couple of one line changes:
>
>     * I think the warning can simply be removed. It's created in a function that tries to spot if a bank of [...]

Agreed. I suppose we could spit out something out at debug if we think it's useful to know that but not required I don't think.

>     * The values u_min and u_max are currently set to 0 and 1 respectively immediately after the warning. Since u is measured in radians, the value of u_max=1 doesn't make much sense. I suggest setting u_max=0. I've tried this and it works OK

Agreed. This would be more correct in line with what the code is saying from the `bin_width` calculation.

martyngigg, 2021-05-28:
Fixed by #31590

PR #31590 (2021-05-27): Remove instrument viewer warning on cylindrical views
**Description of work.**

When viewing instruments that have banks that are width "one detector" in x or y or z, the instrument viewer was giving a warning about "failed to build unwrapped surface" when you viewed in one of the cylindrical projections. Remove the warning and also the code to assign values to u_min and u_max shortly after. The pre-existing values of 0 and 0 are fine for these values which are in radians
Also include some line length changes that come from the latest clang format version

**To test:**

Follow instructions on  [... cut]
closed 2021-05-28 by martyngigg