mantidproject/mantid #31603: Single crystal axis field names in SaveMD

rayosborn, 2021-05-27:
When a MDHistoWorkspace with single crystal axes is saved to a NeXus file, the field names are not compliant with the NeXus standard. In data reduced from a recent Corelli experiment, the NXdata group within the MDHistoWorkspace had the following tree:
```
data:NXdata
  [0,0,5L] = float64(52)
    @frame = 'HKL'
    @long_name = '[0,0,5L]'
    @units = 'r.l.u.'
  [0,K,0] = float64(152)
    @frame = 'HKL'
    @long_name = '[0,K,0]'
    @units = 'r.l.u.'
  [H,0,0] = float64(127)
    @frame = 'HKL'
    @long_name = '[H,0,0]'
    @units = 'r.l.u.'
  errors_squared = float64(51x151x126)
  mask = int8(51x151x126)
  num_events = float64(51x151x126)
  signal = float64(51x151x126)
    @axes = '[0,0,5L]:[0,K,0]:[H,0,0]'
    @signal = 1
```
NeXus field names should be matched by the regular expression `[a-zA-Z0-9_]([a-zA-Z0-9_.]*[a-zA-Z0-9_])?$` (see [here for NeXus naming conventions](https://manual.nexusformat.org/datarules.html#regexpname).  Fields like `[H,0,0]` do not conform. 

I believe that, in the past, the axes were given generic names such as `Q1`, etc, which do conform. The `long_name` attribute, which is currently the same as the new name, can be used if the purpose is to give meaningful axis labels in plots.

rayosborn, 2021-11-24:
The reason it is stale is that nobody ever seems to look at these issues when they are raised.

rayosborn, 2022-05-24:
I repeat my previous comment. Does anyone in the Mantid project actually look at Github issues?

rayosborn, 2022-11-20:
I can see that it's useful to clear out stale issues automatically, but it would be good if it only happened when there was at least one response from a member of the Mantid team.

rayosborn, 2022-11-27:
Would someone on the Mantid team please comment before this is closed.

mducle, 2022-11-28:
@rayosborn @martyngigg I'll look at this when I do the stuff for excitations instrument info for resolution calculations.

mducle, 2023-01-16:
@rayosborn Sorry I can't seem to reproduce this issue. When I tried, I get the axes names as `Q1`, `Q2` etc. 

The code to save the data [is here](https://github.com/mantidproject/mantid/blob/main/Framework/MDAlgorithms/src/SaveMD2.cpp#L137) and uses the `DimensionId` rather than the dimension name. If you run Mantid and open up the workspace information, you can see the dimension ID at the end of the dimension: 

[image]

If I use the following script on a Corelli data file in the Mantid test suite,

```python
from mantid.simpleapi import *

Load(Filename='ExternalData/Testing/Data/UnitTest/CORELLI_29782.nxs', OutputWorkspace='CORELLI_29782')
SetUB(Workspace='CORELLI_29782', a=4, b=4, c=4)
ConvertToMD(InputWorkspace='CORELLI_29782', QDimensions='Q3D', dEAnalysisMode='Elastic', Q3DFrames='HKL', QConversionScales='HKL', OutputWorkspace='CORELLI_29782_hkl')
BinMD(InputWorkspace='CORELLI_29782_hkl', AlignedDim0='[H,0,0],-1,1,50', AlignedDim1='[0,K,0],-1,1,50', AlignedDim2='[0,0,L],-1,1,50', OutputWorkspace='CORELLI_29782_histo')
SaveMD(InputWorkspace='CORELLI_29782_histo', Filename='corelli_29782_histo.nxs')
```

I get the `Q1`, `Q2` etc dimension IDs.

Looking at the saved nxs file in nexpy, I get the same dimension as labels:

[image]

rayosborn, 2023-01-16:
That certainly used to be the way it was handled in the past, as I wrote in my original post. I presume your tests used different algorithms to the ones used to generate our data and I'm pleased if that is still the usual output. Perhaps there are just one or two 'bad' algorithms.

One of the files showing this problem is on the SNS analysis computer at `/SNS/CORELLI/IPTS-26395/shared/OP020_noCC/normData_191434.nxs`. Can you tell from the `MantidAlgorithm` list which one might be causing the problem?

commit 287581de (2023-01-18): Fix #31603

Ensures MD NeXus files have dimension names which adhere to the
NeXus definitions https://manual.nexusformat.org/datarules.html
PR #35005 (2023-01-19): Fix #31603 - Non compliant NeXus dimension IDs
Ensures MD NeXus files have dimension names which adhere to the [NeXus definitions](https://manual.nexusformat.org/datarules.html#regexpname).

**Description of work.**

Some Mantid algorithms (e.g. `MDNorm`) need axes with names in a specific format, e.g. `[H,0,0]`, which is not allowed by the NeXus standard. We have a separate attribute `long_name` which is a free format string where these names can be stored, but currently the `SaveMD2` algorithm also writes the Mantid (non-NeXus-compliant) names as the dimension ID.

This PR changes `SaveMD2`  [... cut]
commit 35a61098 (2023-01-19): Merge pull request #35005 from mantidproject/31603_savemd_axis_names

Fix #31603 - Non compliant NeXus dimension IDs
closed 2023-01-19 by mantid-builder