mantidproject/mantid #31605: ICP_alarm files for INES prevent loading logs

DanielMurphy22, 2021-05-28:
### Expected behavior
Loading a raw file loads the logs successfully.
### Actual behavior
Loading INES raw data from a recent cycle gives the following error 
```
Unable to successfully run LoadLog Child Algorithm:
File/archive/NDXINES/Instrument/data/cycle_20_3/INS17770_ICPalarm.txt is not a standard ISIS log file.
Expected to be a file starting with DateTime String format.
```
### Steps to reproduce the behavior
- Run Load on /archive/NDXINES/Instrument/data/cycle_20_3/INS17770.raw
- Mantid also tries to pick up the ICPalarm.txt file and then cannot load the logs fully.
- In the case of run INS17770, the ICP_alarm file is empty
### Platforms affected
All

SpencerHowells, 2021-05-28:
@DanielMurphy22 It looks as if it is the same problem that we had with old IRIS data. This has files that are not standard. We now just give a warning message & ignore the error messages. The data is loaded ok. In order to find out about the missing data we had to delve into the old data acquisition programs & were unsuccessful in working out how to interpret the data. Fortunately that data is not required.

thomashampson, 2021-06-28:
For some reason this is not happening on Windows.

PR #32012 (2021-07-22): Handle empty alarm log files
**Description of work.**
Added check for empty ICPalarm log files when loading raw data files to prevent LoadLog throwing an exception when it attempts to process an empty log file. The user will no longer see a worrying warning message for a situation that is normal.

**To test:**
Do not test on Windows (see #31986 for reason).
- Change the log level to `information` by right clicking the messages widget
- Run Load on /archive/NDXINES/Instrument/data/cycle_20_3/INS17770.raw
- Verify that there are no warning messages
- Verify that the following information message [... cut]
closed 2021-07-22 by martyngigg