mantidproject/mantid #31623: Windows Clean/Dirty Install Smoke Tests

robertapplin, 2021-06-01:
Before testing:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

Afterwards:
- Comment below with any issues you came across.
- If no issues were found, or they are now all resolved, please close the testing issue.
- Check the master issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :soap: :hankey:

### Dirty install

* Make sure that you have several versions of Mantid installed
 * Last release
 * A nightly
 * If possible an old release
* Install the latest version of the new Mantid
- [ ] Check that Mantid boots up correctly

### Clean install

* Remove all existing Mantid versions and associated files
* Windows:
 * Uninstall the program
 * Clear shortcuts from desktop
 * Clean out the registry
  * Load regedit (Command Prompt > regedit)
  * Find `HKEY_CURRENT_USER > SOFTWARE > Mantid` delete it

**On macOS** :
 * Remove the application
 * Remove the `~/.mantid directory`
 * Remove `~/Library/Preferences/org.mantidproject.MantidPlot.plist`

**On Linux** :
 * Remove the package: `/opt/Mantid`
 * Remove `~/.config/Mantid`
 * Remove `~/.mantid/`

* Re-install the latest version of the new Mantid
- [ ] Check that Mantid boots up correctly

thomashampson, 2021-06-01:
`Find HKEY_CURRENT_USER > SOFTWARE > Mantid delete it`
The key was not there.

thomashampson, 2021-06-01:
>     * Find `HKEY_CURRENT_USER > SOFTWARE > Mantid` delete it
The key doesn't exist for me

closed 2021-06-01 by thomashampson