mantidproject/mantid #31624: Windows Fitting Smoke Tests

robertapplin, 2021-06-01:
Before testing:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

Afterwards:
- Comment below with any issues you came across.
- If no issues were found, or they are now all resolved, please close the testing issue.
- Check the master issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :muscle: :runner:

* Fitting to data [directions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/01_fitting_models_to_data.html#fitting-models-to-data)
- [ ] Simple fit 
- [ ] Activating the fitting tools 
- [ ] Selecting the fitting range 
- [ ] Other Fit Function Settings properties 
- [ ] Setting up a fit model 
- [ ] Adjusting fit function parameter 
- [ ] Saving a model 
- [ ] Tying and constraining fit parameters 
- [ ] Execute your fit 
- [ ] Inspecting a fit result 
* Additional options [directions](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/03_fit_model_choices.html)
- [ ] Non peak model + background 
- [ ] User defined function

RichardWaiteSTFC, 2021-06-01:
When a spectrum/workspace is added or removed from the plot while a fit browser is open, the workspace index and workspace drop-down box aren't refreshed. If you remove a curve from the plot settings you can still select it to fit and this causes a sad mantid/unexpected exception. If you toggle the fit browser off/on it correctly populates the workspace drop-down box.

RichardWaiteSTFC, 2021-06-01:
If you fix a parameter and then change the value and fit it fixes it at the changed value (i.e. not the value at which it was fixed)
[image]

RichardWaiteSTFC, 2021-06-01:
When you tie a parameter to a parameter in another function and then remove that function the tie is still present (though it doesn't cause any crash/exception when you fit).
[image]

RichardWaiteSTFC, 2021-06-01:
Getting unexpectedly large errorbars on fitted curves - I thought this was fixed by #29935?
[image]

RichardWaiteSTFC, 2021-06-01:
> Getting unexpectedly large errorbars on fitted curves - I thought this was fixed by #29935?
> [image]

I ran the test in #29935 and it looked OK - what do you think @StephenSmith25 ?

RichardWaiteSTFC, 2021-06-01:
The scale in the tabulated function appears to be out by an order of magnitude - the two workspaces differ by a scale factor of 1.4
[image]

RichardWaiteSTFC, 2021-06-01:
No other issues that I noticed - other than #31552  about loading a model not applying ties which should be fixed by PR #31600

RichardWaiteSTFC, 2021-06-01:
Put in an issue for not being able to tie a parameter To Function - but it could be that I'm not doing it right!

closed 2021-06-01 by RichardWaiteSTFC