mantidproject/mantid #31626: Windows Plotting Smoke Tests

robertapplin, 2021-06-01:
Before testing:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

Afterwards:
- Comment below with any issues you came across.
- If no issues were found, or they are now all resolved, please close the testing issue.
- Check the master issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :chart_with_upwards_trend: :bar_chart:

* 1D plotting:[instructions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/03_displaying_1D_data.html#displaying-1d-data)
 - [ ] Simple plot 
 - [ ] Another way to plot 
 - [ ] Adding curves to existing plots 
 - [ ] Also, test out [waterfall](https://docs.mantidproject.org/nightly/plotting/WaterfallPlotsHelp.html#waterfall-plots) and [tiled]( https://docs.mantidproject.org/nightly/plotting/1DPlotsHelp.html#tiled-plots)
 - [ ] Check Toolbar buttons

* 2D plotting: [instructions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/04_displaying_2D_data.html#displaying-2d-data)
 - [ ] Plot all spectra 
 - [ ] Change colour map 
 - [ ] [Contour plot](https://docs.mantidproject.org/nightly/plotting/ColorfillPlotsHelp.html#contour-plots) (under 3D menu) 
 - [ ] Check Toolbar buttons

* [3D plotting](https://docs.mantidproject.org/nightly/plotting/3DPlotsHelp.html):
 - Load some data eg `LOQ7041` from the ISIS sample data
 - [ ] 3D surface
 - [ ] 3D wire frame  
 - [ ] Check Toolbar buttons

 ## Sliceviewer
 - [ ] Overly long instructions (don't spend hours!) and data [here](https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html). In particular try editing the data in a workspace while it is open in Sliceviewer! ([step 10](https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html#alter-the-underlying-workspace))

sf1919, 2021-06-01:
Having problems with ADS.retrieve when opening scripts I've saved or scripts saved to clipboard. I get a file does not exist error and I'm using one of the Training Data files (MAR11060). I've checked and have Search Data Archive set as 'default facility only - ISIS' in Manage User Directories. Is it just that the file in question isn't in the ADS?

sf1919, 2021-06-01:
The colour contour/wireframe option is not available on the toolbar of a wireframe
[image]

sf1919, 2021-06-01:
In the Wireframe images (https://docs.mantidproject.org/nightly/plotting/3DPlotsHelp.html) there is a Wireframe plot for MAR11060. When loading this Fit is missing from the toolbar

What I'm getting:-
[image]

What the image from the guide suggests I should be getting:-
https://docs.mantidproject.org/nightly/_images/PlotClickMenus3DWireframe.png

robertapplin, 2021-06-01:
> Having problems with ADS.retrieve when opening scripts I've saved or scripts saved to clipboard. I get a file does not exist error and I'm using one of the Training Data files (MAR11060). I've [...]

You might need to set a Data Search Directory in the `Manage User Directories` dialog. Try `Browse To Directory` to the location of MAR11060 and add it

sf1919, 2021-06-01:
> You might need to set a Data Search Directory in the `Manage User Directories` dialog. Try `Browse To Directory` to the location of MAR11060 and add it

I already have the Training Course Data as one of the directories set for it to search so it doesn't seem to be that.

robertapplin, 2021-06-01:
> 
> I already have the Training Course Data as one of the directories set for it to search so it doesn't seem to be that.

Ah that is strange, does it have the exact same name in the training data directory? If its a raw or nxs file could you try putting .raw or .nxs on the end of the name in the python script?

sf1919, 2021-06-01:
> Ah that is strange, does it have the exact same name in the training data directory? If its a raw or nxs file could you try putting .raw or .nxs on the end of the name in the python script?

That doesn't seem to work either. The problem is that the script is being generated as ```MAR11060 = ADS.retrieve('MAR11060')``` which just isn't working for me. Could you try and recreate it? Need to load ``MAR111969``` and then create some plots, generate a script and try and run that script?

sf1919, 2021-06-01:
Following the instructions on https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html for SliceViewer testing I'm trying to use SliceViewer on ```CNCS_7860_event.nxs``` and follow the testing instructions. Double clicking the axes doesn't seem to bring up any dialog boxes to enable it to be changed (is that correct?). 

Also trying to change X and Y as suggested throws a Runtime error. Apprently the ```indexOfValue``` is out of range. In fact it keeps generating this error every so often without me doing anything more than having chaged the axis. Is it the data rather than me?

sf1919, 2021-06-01:
I'm getting error messages when trying to overlay peaks in SliceViewer when following https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html. When overlaying peaks I get an error message but the peaks table does appear. Same occurs when trying to overlay the integrated_peaks.

sf1919, 2021-06-01:
I've finished the testing and I've not found anything else. Given the issues I've had understanding the SliceViewer testing instructions (and not just this time) I think its likely that the X/Y Runtime error I'm getting is user error rather than something real. Meanwhile the confusion with the documentation for the toolbar for wirefranes is fairly minor. As a result I haven't raise either of these as separate issues.

The issue with peak fitting is signficant and has been raised.

closed 2021-06-01 by sf1919

robertapplin, 2021-06-01:
Thanks for testing this. Were you able to replicate the X/Y runtime error? If so it might be worth raising an issue for it as we should try to prevent these types of errors. If you think its a minor problem we can set the milestone to Release 6.2

sf1919, 2021-06-01:
I've gone through all my workspaces again and cannot replicate the X/Y runtime error again so I definitely think it was user error!