mantidproject/mantid #31628: Windows Workspace/Algorithm Smoke Tests

robertapplin, 2021-06-01:
Before testing:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

Afterwards:
- Comment below with any issues you came across.
- If no issues were found, or they are now all resolved, please close the testing issue.
- Check the master issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :man_astronaut: :woman_astronaut: :space_invader:

* Load some workspaces, try 5 from different instruments
* Show detectors, check Instrument Viewer, follow  directions [in the documentation](https://www.mantidproject.org/MantidPlot:_Instrument_View)
  - [ ] Render tab 
  - [ ] Draw tab 
  - [ ] Pick tab 
  - [ ] Instrument tree tab 
* Test 5 - 10 random algorithms, follow [the documentation](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/algorithms_workspaces_and_histories/02_algorithms.html#algorithms) and [pick your favourite](https://docs.mantidproject.org/v3.9.0/algorithms/)
 - [ ] Help button 
 - [ ] Validation of inputs 
 - [ ] Overwriting workspace 
 - [ ] Run the algorithm 
- [ ] Check right-click options of the workspaces, they are listed [in the documentation](https://www.mantidproject.org/MantidPlot:_The_Workspace_Menu#Save_Nexus).

DannyHindson, 2021-06-01:
The https://www.mantidproject.org/MantidPlot:_Instrument_View link doesn't work. It should perhaps be https://www.mantidproject.org/MantidPlot__Working_with_Instruments.html works instead - although possible the MantidPlot link won't last much longer either

robertapplin, 2021-06-01:
Thanks, I'll update this link for next time

robertapplin, 2021-06-01:
> The https://www.mantidproject.org/MantidPlot:_Instrument_View link doesn't work. It should perhaps be https://www.mantidproject.org/MantidPlot__Working_with_Instruments.html works instead - although possible the MantidPlot link won't last much longer either

I tried these links a few hours ago and I found the same as you, the first one does not work. However, I have just tried them again and now I find that the first link works but the second one doesn't. Has this changed for you too? 😕

DannyHindson, 2021-06-01:
> I tried these links a few hours ago and I found the same as you, the first one does not work. However, I have just tried them again and now I find that the first link works but the second one [...]

Yes the original works for me now. I think I heard David mention that he'd found a problem and fixed it while I was on Gather earlier so I suspect that's what it is

DannyHindson, 2021-06-01:
Summary of my findings from the tests - not much to report. All works fine apart from two points which are not big issues I think:
- the show detectors screen has some new columns to show the calibration (diffractometer constants) for diffraction instrument. These are only shown if Mantid thinks the workspace is "elastic". At the moment this includes SANS and Muon instruments. I think the new columns probably aren't of interest to these users so ideally the filter should be tightened
- the show detectors screen doesn't show any information for MARI datasets for me. I see the same in 6.0 though so doesn't seem to be a new issue and quite possibly MARI specific

robertapplin, 2021-06-02:
Thanks @DannyHindson , sounds like only very minor issues not for this release 👍

closed 2021-06-02 by robertapplin