mantidproject/mantid #31631: MacOS Fitting Smoke Tests

robertapplin, 2021-06-01:
Before testing:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

Afterwards:
- Comment below with any issues you came across.
- If no issues were found, or they are now all resolved, please close the testing issue.
- Check the master issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :muscle: :runner:

* Fitting to data [directions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/01_fitting_models_to_data.html#fitting-models-to-data)
- [ ] Simple fit 
- [ ] Activating the fitting tools 
- [ ] Selecting the fitting range 
- [ ] Other Fit Function Settings properties 
- [ ] Setting up a fit model 
- [ ] Adjusting fit function parameter 
- [ ] Saving a model 
- [ ] Tying and constraining fit parameters 
- [ ] Execute your fit 
- [ ] Inspecting a fit result 
* Additional options [directions](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/03_fit_model_choices.html)
- [ ] Non peak model + background 
- [ ] User defined function

DanielMurphy22, 2021-06-01:
Testing on Big Sur. Also, recreated the issue @RichardWaiteSTFC found where changing the value of a parameter away from the tied value, fits incorrectly to the changed value.

DanielMurphy22, 2021-06-01:
Also, found that on Big Sur the error reporter doesn't launch properly!

DanielMurphy22, 2021-06-01:
No other issues found. Opened : https://github.com/mantidproject/mantid/issues/31658

closed 2021-06-01 by DanielMurphy22