mantidproject/mantid #31634: MacOS Python Smoke Tests

robertapplin, 2021-06-01:
Before testing:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

Afterwards:
- Comment below with any issues you came across.
- If no issues were found, or they are now all resolved, please close the testing issue.
- Check the master issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :snake:

* Test that the Python scripting window works, [directions here](https://docs.mantidproject.org/nightly/workbench/scriptwindow.html)
 - [ ] Editor options 
 - [ ] Execution options 
 - [ ] Script output  
- [ ] Perform some workspace algebra 
- [ ] Test numpy functionality 
- [ ] Use the scripting window to run some scripts 
- [ ] Run through some examples from [the documentation](http://www.mantidproject.org/Python_In_Mantid), 3 or 4 examples is enough

DanielMurphy22, 2021-06-01:
The from future import line should be removed from https://docs.mantidproject.org/nightly/workbench/scriptwindow.html as we are on Python3 now!

DanielMurphy22, 2021-06-01:
Not directly linked to this, but the exercise in Python in Mantid has AlignDetectors in it, so needs updating: https://docs.mantidproject.org/nightly/tutorials/python_in_mantid/solutions/01_pim_sol.html#pim-sol

DanielMurphy22, 2021-06-02:
No other issues found