mantidproject/mantid #31648: Ubuntu Python Smoke Tests

robertapplin, 2021-06-01:
* Follow the instructions [here](https://download.mantidproject.org/ubuntu.html) to install the .deb for Ubuntu

Before testing:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

Afterwards:
- Comment below with any issues you came across.
- If no issues were found, or they are now all resolved, please close the testing issue.
- Check the master issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :snake:

* Test that the Python scripting window works, [directions here](https://docs.mantidproject.org/nightly/workbench/scriptwindow.html)
 - [ ] Editor options 
 - [ ] Execution options 
 - [ ] Script output  
- [ ] Perform some workspace algebra 
- [ ] Test numpy functionality 
- [ ] Use the scripting window to run some scripts 
- [ ] Run through some examples from [the documentation](http://www.mantidproject.org/Python_In_Mantid), 3 or 4 examples is enough

DanielMurphy22, 2021-06-02:
Saving a script that contains the following, and reopening it, Mantid asks and then puts the simple.api import line BELOW the Load command. This only occurs for a one line script with a blank line after it.
```
ws = Load('MAR11060')

```

DanielMurphy22, 2021-06-02:
Issues found are not worth fixing. Testing on Ubuntu 18.04

closed 2021-06-02 by DanielMurphy22