mantidproject/mantid #31649: Ubuntu Workspace/Algorithm Smoke Tests

robertapplin, 2021-06-01:
* Follow the instructions [here](https://download.mantidproject.org/ubuntu.html) to install the .deb for Ubuntu

Before testing:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

Afterwards:
- Comment below with any issues you came across.
- If no issues were found, or they are now all resolved, please close the testing issue.
- Check the master issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :man_astronaut: :woman_astronaut: :space_invader:

* Load some workspaces, try 5 from different instruments
* Show detectors, check Instrument Viewer, follow  directions [in the documentation](https://www.mantidproject.org/MantidPlot:_Instrument_View)
  - [ ] Render tab 
  - [ ] Draw tab 
  - [ ] Pick tab 
  - [ ] Instrument tree tab 
* Test 5 - 10 random algorithms, follow [the documentation](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/algorithms_workspaces_and_histories/02_algorithms.html#algorithms) and [pick your favourite](https://docs.mantidproject.org/v3.9.0/algorithms/)
 - [ ] Help button 
 - [ ] Validation of inputs 
 - [ ] Overwriting workspace 
 - [ ] Run the algorithm 
- [ ] Check right-click options of the workspaces, they are listed [in the documentation](https://www.mantidproject.org/MantidPlot:_The_Workspace_Menu#Save_Nexus).

MathieuTillet, 2021-06-01:
In the instrument viewer, if one save an image, change the axis view, then save another image, the new image will show the previous view, before the axis change. I don't think it's a particularly important bug, but nonetheless, I don't think it is meant to work this way either.

MathieuTillet, 2021-06-01:
Apart from this nitpick, no complain, so I close the issue.

closed 2021-06-01 by MathieuTillet