mantidproject/mantid #31660: Fixed parameter in fit browser fixes at current value not the value at which it was initially fixed

RichardWaiteSTFC, 2021-06-01:
### Expected behavior
If you fix a parameter and then change the value, when you fit it be fixed at the value you initially set

### Actual behavior
It fixes at the current value
[image]

### Steps to reproduce the behavior
(1) Make some data e.g. 
`ws = CreateSampleWorkspace(Function="User Defined",   UserDefinedFunction="name=LinearBackground, A0=0.3;name=Gaussian, PeakCentre=5, Height=10, Sigma=0.7",  NumBanks=1, BankPixelWidth=1, XMin=0, XMax=10,  BinWidth=0.1, WorkspaceType="Event")`
(2) Open fit browser
(3) Add some function (e.g. Gaussian + FlatBackground) 
(4) Fix a parameter (e.g. A0=0)
(5) Change the value of that parameter (e.g. A0 -> 100)
(6) Fit

### Platforms affected
Found smoke testing fitting on windows #31624

thomashampson, 2021-09-28:
I tyhink this behaviour is fine, except that the value in the `Tie` box should track the value of the parameter.

sf1919, 2023-02-08:
I am unable to recreate this. I suspect is has been fixed by either #34817 or #35081