mantidproject/mantid #31665: Peak overlay error message in SliceViewer

sf1919, 2021-06-01:
### Expected behavior
Following the instructions in https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html to create workspaces and create peak workspaces you should be able to overlay peaks on ```histoWS_3D``` and ``md_4D```

### Actual behavior
When trying to overlay either peaks or integrated peaks an error message occurs. The Table does load but the peaks are not overlaid.

```File "C:\MantidInstall\bin\mantidqt\widgets\sliceviewer\view.py", line 617, in peaks_overlay_clicked
   self.presenter.overlay_peaks_workspaces()
 File "C:\MantidInstall\bin\mantidqt\widgets\sliceviewer\presenter.py", line 362, in overlay_peaks_workspaces
   self._create_peaks_presenter_if_necessary().overlay_peaksworkspaces(names_to_overlay)
 File "C:\MantidInstall\bin\mantidqt\widgets\sliceviewer\peaksviewer\presenter.py", line 218, in overlay_peaksworkspaces
   self.notify(PeaksViewerPresenter.Event.OverlayPeaks)
 File "C:\MantidInstall\bin\mantidqt\widgets\sliceviewer\peaksviewer\presenter.py", line 250, in notify
   presenter.notify(event)
 File "C:\MantidInstall\bin\mantidqt\widgets\sliceviewer\peaksviewer\presenter.py", line 88, in notify
   self._overlay_peaks()
 File "C:\MantidInstall\bin\mantidqt\widgets\sliceviewer\peaksviewer\presenter.py", line 112, in _overlay_peaks
   self.model.draw_peaks(self._view.sliceinfo, self._view.painter)
 File "C:\MantidInstall\bin\mantidqt\widgets\sliceviewer\peaksviewer\model.py", line 77, in draw_peaks
   peak_repr = draw_peak_representation(peak_origin, peak.getPeakShape(), slice_info,
 File "C:\MantidInstall\bin\mantidqt\widgets\sliceviewer\peaksviewer\representation\draw.py", line 49, in draw_peak_representation
   return _get_factory(peak_shape).draw(peak_origin, peak_shape, slice_info, painter, fg_color,
 File "C:\MantidInstall\bin\mantidqt\widgets\sliceviewer\peaksviewer\representation\ellipsoid.py", line 44, in draw
   peak_origin = slice_info.transform(peak_origin)
 File "C:\MantidInstall\bin\mantidqt\widgets\sliceviewer\sliceinfo.py", line 87, in transform
   point[self._display_z]))
Boost.Python.ArgumentError: Python argument types in
   None.__getitem__(V3D, NoneType)
did not match C++ signature:
   __getitem__(class Mantid::Kernel::V3D {lvalue} self, int index)```

### Steps to reproduce the behavior
- Follow the instructions in https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html to create workspaces and create peak workspaces
- Open SliceViewer on  ```histoWS_3D``` workspace
- Click peak overlay button
- choose either peaks or integrated_peaks

### Platforms affected
Windows 10

RichardWaiteSTFC, 2021-06-01:
It is not possible to overlay the integrated ellipsoid peaks on a 4D workspace - can you confirm that you get this error in 3D as well?

sf1919, 2021-06-01:
I can confirm it is happening on ```histoWS_3D``` which I thought was the 3D workspace

RichardWaiteSTFC, 2021-06-01:
The instructions say to make peaks by picking them on the instrument viewer of some SXD data - were you overlaying these SXD peaks onto the faked MD data?

sf1919, 2021-06-01:
Yes

RichardWaiteSTFC, 2021-06-01:
> Yes

Maybe it's to do with that then, I'm trying to recreate the issue but I ran into an issue loading the SXD data  from my debug build (#31667) so am currently rebuilding the whole project in release!

sf1919, 2021-06-01:
This is no other way to do the testing and it has worked before in the past.

RichardWaiteSTFC, 2021-06-01:
> This is no other way to do the testing and it has worked before in the past.

Yes - it does look like you've found a problem and it may or may not impact more realistic usage! I would say the instructions are actually for overlaying the peaks on the SXD data, however trying to overlay them on the other MD workspaces is a good thing to test as well.

sf1919, 2021-06-01:
> Yes - it does look like you've found a problem and it may or may not impact more realistic usage! I would say the instructions are actually for overlaying the peaks on the SXD data, however trying to overlay them on the other MD workspaces is a good thing to test as well.

When I open the SXD23767 file in SliceViewer the peak overlay button is greyed out I'm now very confused!

RichardWaiteSTFC, 2021-06-01:
> When I open the SXD23767 file in SliceViewer the peak overlay button is greyed out. I'm now very confused!

You have to overlay the peaks on an MD object (`md_non_ortho` in the instructions)

sf1919, 2021-06-01:
That works no problems! Maybe it really is a problem with how the testing has been written as there is nothing to suggest you shouldn't do all the tests for each workspace you created on the top half of the page!

martyngigg, 2021-06-01:
Dropping from the release as it's mostly a documentation issue that a user would rarely hit.

RichardWaiteSTFC, 2021-06-01:
The error seems to be related to the fact the dimensions of the fake datasets are not interpreted as being Q (this means the sliceinfo ends up being invalid, it gets treated like a matrix workspace (i.e. 2D) so sliceinfo.display_z == None). All the dimensions in the `md_non_ortho`  are Q and everything works as expected. So this looks like a problem with how we make the fake dataset! If we make it like this there is no error.
```
ws = CreateMDWorkspace(Dimensions='3', Extents='-5,5,-4,4,-3,3',
                       Names='H,K,L', Units='r.l.u.,r.l.u.,r.l.u.',
                       Frames='HKL,HKL,HKL',
                       SplitInto='2', SplitThreshold='50')
```

RichardWaiteSTFC, 2021-06-02:
I will update the test code in the docs with the following which will make valid 4D MDEvent and 3D MDHisto workspaces
```
md_4D = CreateMDWorkspace(Dimensions=4, Extents=[-1,1,-1,1,-1,1,-10,10], 
    Names="H,K,L,E", Frames='HKL,HKL,HKL,General Frame',Units='r.l.u.,r.l.u.,r.l.u.,meV')
histoWS_3D = BinMD(InputWorkspace=md_4D, AlignedDim0='H,-1,1,100', AlignedDim1='K,-1,1,100', AlignedDim2='L,-1,1,100') 
print([histoWS_3D.getDimension(n).getMDFrame().isQ() for n in range(histoWS_3D.getNumDims())]) # check is Q dimension
```
I have checked and no error is thrown when overlaying peaks on these workspaces (although the peak functionality is only really supported for 3D).

PR #31683 (2021-06-02): Update sliceviewer testing docs
**Description of work.**

Updated sliceviewer testing docs to make it clearer which functionality should be tested on the different workspace types. 
In addition I have fixed a bug in the code that made the fake md workspaces such that the dimensions are now recognised as being Q (this fixes issue #31565 which arose because the sliceinfo assumed the workspace was 2D MatrixWorkspace).

**To test:**

(1) Code review
(2) Build docs without error
(3) Read html produced

Fixes #31665 

*This does not require release notes* because **it is a change to documentation**
 [... cut]
closed 2021-06-02 by gemmaguest