mantidproject/mantid #31687: EngDiff UI crashes on IDAaaS when removing runs

RichardWaiteSTFC, 2021-06-02:
UI crashes when you remove runs and if you subsequently try to load those same runs again in the same UI window 

### Platforms affected
IDAaaS

DannyHindson, 2021-07-13:
@RichardWaiteSTFC - do you remember which tab in the UI this crash related to? Was it removing and reloading files on the Fitting tab using the "Remove All" and "Load" buttons?

RichardWaiteSTFC, 2021-07-13:
> @RichardWaiteSTFC - do you remember which tab in the UI this crash related to? Was it removing and reloading files on the Fitting tab using the "Remove All" and "Load" buttons?

It was the fitting tab, using the remove selected or remove all and then loading the same runs - the bug for remove all might have been fixed by #32000?

DannyHindson, 2021-07-14:
I've done some tests with a build before\after the changes in #32000 and I think that's fixed it so I'll close this issue. I was able to reproduce a crash before the changes by:

a) loading some data onto the fitting tab
b) hitting clear in the workspace widget (this clears the table in the Fitting tab as well as the main ws list)
c) hitting "Remove All" back on the fitting tab

This gave the following error that Tom's changes fix:

[image]

closed 2021-07-14 by DannyHindson