mantidproject/mantid #31696: Peak finding machine learning - WISH

DannyHindson, 2021-06-02:
Discuss progress to date with Pascal

RichardWaiteSTFC, 2021-06-22:
Following a meeting with Pas and Kuangdai Leng - the algorithm is in python and requires the following libraries: multiprocessing, skimage, sklearn. It operates on a 3D array with the x-axis being lambda and the other dimensions being the detector banks folded out in 2D like in instrument view - making this quite general. Currently the data are not normalised to monitor before-hand (introduces noise).
The algorithm is as follows
Separates the 3D array into blocks of dimensions chosen by the user (with some overlap between boxes)
In each box (looped over in parallel using multiprocessing library) uses 2D blob detection in skimage for the data projected along each of the 3 axes (i.e. performs 3 x 2D convolutions of some kernel, e.g. mexican hat func. - a bit like in nsxtools)
Identifies blobs in 2D over some threshold and merges them is above some threshold of area overlap
From 2D blobs in the 3 images identify peaks in 3D (two options: to use clustering in sklearn to find n peaks or just take strongest peak).
Check for overlap of peaks between blocks
It seems to work quite nicely, there are ~5 options the user can change (which is fewer than i expected) - for all the rest it seems like good defaults have been chosen and I see no reason why it wouldn't work for other instruments.
The code we would put in mantid consists of 3 files: the core algorithm, helper functions, user input file/script.
I think for now this could be used as a standalone code as long as mantid/IDAaaS could have the above dependencies - slightly longer term we could wrap it in an algorithm. They are currently writing a paper on it so they will still be working on it and maintaining it for a bit, but very long term I don't think it will be maintained unless we pick it up (there may be some licensing/copyright stuff to consider?).
In term interaction with mantid I see two obvious next steps:
Convert output peaks as indices in 3D array to a PeakWorkspace (easy to hard-code mapping from index->detid in WISH, then just add peak based on detid and TOF)
Produce 3D array from MatrixWorkspace in a general way - in SXD we might want to have a 3D array per bank because the banks separation is such that a peak won't spill over into another bank. On WISH we might consider each half of the instrument. It will very much depend on the instrument.

Note the machine learning is inside the sk-library, this algorithm was not trained on any WISH specific data (it is unsupervised)

RichardWaiteSTFC, 2021-06-25:
Closing this - we can decide what to do after the meeting on 30/06

closed 2021-06-25 by RichardWaiteSTFC