mantidproject/mantid #31705: Implement output individual function fits in other IDA fit tabs

Harrietbrown, 2021-06-03:
**Original reporter:** @SpencerHowells

In the convfit tab in IDA there is an option to Output Composite Members. This option should be available in other tabs. This option also is disabled when the fitting type is made simultainious and should be made to work for that case.

SpencerHowells, 2021-06-03:
@Harrietbrown ConvFit should have Output Composite Members & Convolve Members. IqtFit just Output Composite Members. Other 2 neither.

SpencerHowells, 2021-11-30:
@Harrietbrown Keep this opne please

SpencerHowells, 2022-05-30:
@Harrietbrown keep open please

Harrietbrown, 2023-01-24:
[image] [Implement output individual function fits in other IDA fit tabs](https://trello.com/c/iidtNsIX/15-implement-output-individual-function-fits-in-other-ida-fit-tabs)

SpencerHowells, 2024-06-26:
This should only be added to the IqtFit gui. Not needed in MSD or F(Q)

PR #37677 (2024-07-26): Add output composite members option to Iqt
### Description of work
This PR adds `Output Composite Members` to the IQT fitting tab. 
#### Summary of work

Fixes #31705. 

### To test:
1. Go to `Interfaces->Inelastic->I(Q, T)`
2. Click `Add Workspace` and add [irs26176_graphite002_iqt.zip](https://github.com/user-attachments/files/16331874/irs26176_graphite002_iqt.zip)
3. Set the indices from 0-5
4. Add two Expontatials 
5. Click `Run`
6. Plot the output workspaces spectra suffixed with `_Workspace` and check that they have `Data`, `Calc` and `Diff` labels
7. Go back to the interface and mark `O [... cut]
closed 2024-07-26 by robertapplin