mantidproject/mantid #31718: Bug in InstrumentViewManager

gui-co, 2021-06-04:
The instrument view manager:
https://github.com/mantidproject/mantid/blob/b23c947d84794dd60c9d5d2c244502b4437aa6d2/qt/python/mantidqt/widgets/instrumentview/presenter.py#L94

maintains an internal dictionary of instrument viewer instances. This dictionary is not properly cleared and old references of closed instrument viewers are kept. You can observe this behavior with this script:

```python
from mantid.simpleapi import *
from mantidqt.widgets.instrumentview.presenter import InstrumentViewManager
from mantidqt.widgets.instrumentview.api import get_instrumentview

ws = CreateSampleWorkspace()
# open and close instrument viewer
i = get_instrumentview(ws)
i.show_view()
i.container.close() # or use the close button

# print the manager internal dictionary
print(InstrumentViewManager.view_dict)
```

I did not find any call to this manager in mantid code but maybe it is used in user's script.
This has been introduced with #29904

thomashampson, 2021-12-13:
This sample script now causes Mantid to hang. If you try it in on a Windows debug build, the following message is displayed:
[image]

ckendrick, 2022-01-21:
This sample script should now no longer cause a hang after the fix in #33349. The instrument view was being returned before it was fully initialized by the background thread.

@thomashampson We can probably close this issue now unless this problem is still happening

thomashampson, 2022-01-21:
> This sample script should now no longer cause a hang after the fix in #33349. The instrument view was being returned before it was fully initialized by the background thread.
> 
> @thomashampson We can probably close this issue now unless this problem is still happening

@ckendrick I'll double check my Windows build on Monday morning 👍

martyngigg, 2022-02-03:
Fixed by #33015

PR #33015 (2022-02-02): Add test for instrument view manager
**Description of work.**
Added a test to verify the resolution of a bug that was casuing a memory leak when closing the instrument viewer window. The bug was resolved as a part of #32680. I've added release notes because this is an issue that some users were encountering, and it wasn't mentioned explicitly in the PR that fixed the issue.

**To test:**
To test the original bug fix:
1. Load a workspace
2. Right click and select "Show Instrument"
3. Close the instrument viewer using the GUI's close button
4. Run this script from the editor:
```
from mantid.sim [... cut]
closed 2022-02-03 by martyngigg