mantidproject/mantid #31726: Save ceria calibration files in EngDiffUI

clayton-tom, 2021-06-07:
As a user, I want the focussed and normalised ceria data to be saved in the same manner as the pre-calibration data for a vanadium run, to save time by not having to wait for these calculations to be repeated when using the same ceria run again.

RichardWaiteSTFC, 2021-06-29:
@clayton-tom is this part of /related to issue #31904?

clayton-tom, 2021-07-09:
> @clayton-tom is this part of /related to issue #31904?

This wasn't addressed as part of that, would recommend doing this in conjunction with moving the region of interest selection to the focusing stage if that's something Saurabh still wants, should save work on conflicting code in the calibration workflow 👍