mantidproject/mantid #31777: WISH memory leak related to Instrument viewer

DanielMurphy22, 2021-06-14:
**Original reporter:** Pascal / ISIS

### Expected behavior
Replacing workspaces (say by reloading) should not use extra unneccessary memory
### Actual behavior
This can happen, most noticeably with the large workspaces output by WISH, and it is enhanced by instrument viewer.
### Steps to reproduce the behavior

Script:
```
from mantid.simpleapi import *
Load(Filename='path on archive to ... /WISH00044730.nxs', OutputWorkspace='WISH00044730')
CropWorkspace(InputWorkspace='WISH00044730', OutputWorkspace='WISH00044730', XMin=6000, XMax=99900)
NormaliseByCurrent(InputWorkspace='WISH00044730', OutputWorkspace='WISH00044730')
ConvertUnits(InputWorkspace='WISH00044730', OutputWorkspace='WISH00044730', Target='dSpacing')
```
0. Run the above script 3 or 4 times and see if the memory after each time the script completes increases. I was only seeing small increase of ~0.2 GB at most each time if any.
1. Run the above script, then open instrument viewer by right-clicking on the workspace WISH00044730 and selecting "Show Instrument",. Then rerun the script 2 or 3 times. Initially the memory of the workspace is about 17.5 GB, but this climbs quickly.
2. More simply I ran just the Load line and open instrument viewer before rerunning the load a few times
3. Run the Load command, then open AND close instrument viewer. Run the laod command a few times.
4. You can also experiment with clearing (find the Clear button at the top of the Workspaces Toolbox) and then reloading.
5. Oddly after triggering this with Instrument viewer, relaunching Workbench and siply running the 5 line script still saw huge jumps so I'm really not too certain how well insttrument viewer is connected
### Platforms affected
Found on IDAaaS Workbench Nightly and release 6.1

martyngigg, 2021-06-15:
Oh no this is has come back. I thought our switch to jemalloc had sorted this...

thomashampson, 2021-07-29:
I've run the script 100 times and produced a plot, using the following python script on IDAaaS:

<details>
<summary>memory_plot_script.py</summary>
```
import mantid
from mantid.kernel import MemoryStats
from mantid.api import FrameworkManager
from mantid.simpleapi import *
from time import sleep
import matplotlib.pyplot as plt
import numpy as np

   
def process_data(filename):
    Load(Filename=filename, OutputWorkspace='outputWS')
    CropWorkspace(InputWorkspace='outputWS', OutputWorkspace='outputWS', XMin=6000, XMax=99900)
    NormaliseByCurrent(InputWorkspace='outputWS', OutputWorkspace='outputWS')
    ConvertUnits(InputWorkspace='outputWS', OutputWorkspace='outputWS', Target='dSpacing')

def plot_stats(filename, n_runs, clear_ws=False):
    mem = MemoryStats()
    y = []
    fig, ax = plt.subplots()
    plt.show()
    
    for run in range(1, n_runs + 1):
        process_data(filename)
        sleep(10)
        mem.update()
        res = mem.residentMem() / (1024.0 * 1024.0)
        y.append(res)
        x = np.linspace(1, run, run)
        
        # fit a linear curve an estimate its y-values
        a, b = np.polyfit(x, y, deg=1)
        y_est = a * x + b

        ax.clear()
        ax.plot(x, y_est, '-')
        ax.plot(x, y, 'o', color='tab:brown')        
        ax.set_xlabel('run number')
        ax.set_ylabel('memory usage (GB)')
        plt.pause(0.05)

filename='/archive/NDXWISH/Instrument/data/cycle_18_4/WISH00044730.nxs'

plot_stats(filename, 100)
print("Done!")

```
</details>

resulting in:

[image]

DanielMurphy22, 2021-07-30:
Did you open the instrument viewer before running the script?? I think the team in ORNL added a command to open instrument viewer from python.

martyngigg, 2021-08-23:
Tom, should we close this one as we think there is no issue or do we think there are further things to investigate.

thomashampson, 2021-08-23:
I have some other plots from a script that opens and closes the instrument viewer between loads, but I was watiing for IDAaaS to come back online to retrieve them.

gvardany, 2021-09-28:
We've also noticed that https://github.com/mantidproject/mantid/blob/master/qt/python/mantidqt/mantidqt/widgets/instrumentview/presenter.py#L122 is never run, when closing the instrument viewer, so the manager keeps accumulating and never gets cleaned up. If the manager can hold the workspace alive, this is a threat.

gvardany, 2021-09-28:
I meant this is not executed when closing the instrument viewer: https://github.com/mantidproject/mantid/blob/master/qt/python/mantidqt/mantidqt/widgets/instrumentview/presenter.py#L79

martyngigg, 2021-09-28:
It's really difficult to see what is going on there as the close behaviour is defined all over the place.

gvardany, 2021-09-28:
This is the issue we created when we hit upon it: https://github.com/mantidproject/mantid/issues/31718
The fix went to a long feature branch that is in progress. Could you cherry-pick these and see if this helps?
e3e29de
44ec9fe

DavidFair, 2021-09-28:
Profiling on OSX I think the memory leak is unrelated to inst viewer:
[image]

From my screenshot

1 - Load file then delete from ADS, no inst viewer
2 - Load file with new name, open in slice viewer, delete workspace, notice allocation climbs
3 - Open / Close inst viewer then delete workspace, notice drop in allocation.

Either something common to both Inst Viewer and Slice Viewer is leaking or malloc is requesting and releasing blocks of memory in a way that is misleading.

I'm profiling on MSVC at the moment, as the tools there tend to be better (at least for me) capturing stacks / heaps with snapshots

DavidFair, 2021-09-28:
So MSVC is much clearer what's going on, ignore the OSX trace as Malloc was being lazy

- We correctly alloc / release memory with load / delete
- The above commits fix the memory leak for MSVC

martyngigg, 2021-09-28:
> This is the issue we created when we hit upon it: #31718 The fix went to a long feature branch that is in progress. Could you cherry-pick these and see if this helps? [...]

It's good that we have found the cause here. Do those changes overlap somewhat with the base `ObservingView` behaviour and is it muddying what is going on there? For example, I can see a `close_signal` is already defined?

RichardWaiteSTFC, 2021-09-28:
> Sliceviewer leaks memory even with #32642 so that needs a similar (I'm guessing) fix too

Yes I think WISH have observed a similar memory leak, but they use both sliceviewer and instrument view - the latter is more important for them because they run out of RAM as the Workspace2Ds are bigger than the MD workspaces they make.

gvardany, 2021-09-29:
> It's good that we have found the cause here. Do those changes overlap somewhat with the base ObservingView behaviour and is it muddying what is going on there? For example, I can see a close_signal is already defined?
@gui-co Could you provide more detail as you have found the problem?

thomashampson, 2021-11-19:
@gvardany @gui-co I've created a PR  #33015 that adds a test to verify that this has been fixed (basically the same as the script in the issue @gui-co created). It turns out that it was fixed as part of another PR recently, which will probably cause conflicts with the commits on your development branch.

Both this issue and #31718 will close once my PR with the test in is merged.

martyngigg, 2022-02-02:
Fixed by #33015

PR #33015 (2022-02-02): Add test for instrument view manager
**Description of work.**
Added a test to verify the resolution of a bug that was casuing a memory leak when closing the instrument viewer window. The bug was resolved as a part of #32680. I've added release notes because this is an issue that some users were encountering, and it wasn't mentioned explicitly in the PR that fixed the issue.

**To test:**
To test the original bug fix:
1. Load a workspace
2. Right click and select "Show Instrument"
3. Close the instrument viewer using the GUI's close button
4. Run this script from the editor:
```
from mantid.sim [... cut]
closed 2022-02-02 by martyngigg