mantidproject/mantid #31820: Workspaces interference in python scripts

abuts, 2021-06-17:
### Expected behavior
The script below is expected to produce and plot two identical workspaces. 
```Python
# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np
from mantid.api import AnalysisDataService as ADS

def plot_ws(ws_name,fig_num,max_range):
    ConvertSpectrumAxis(InputWorkspace=ws_name,OutputWorkspace='_PWS',Target='Theta')
    Transpose(InputWorkspace='_PWS',OutputWorkspace=ws_name+'_PWS')

    pws = ADS.retrieve(ws_name+'_PWS')

    fig = plt.figure(fig_num)
    axes = fig.add_subplot(111, projection="mantid")
    #fig, axes = plt.subplots(num=1, subplot_kw={'projection': 'mantid'})
    axes.plot(pws, color='#1f77b4', label=ws_name, linestyle='None', marker='.', markersize=1.0, wkspIndex=0, zorder=2.1)
    axes.tick_params(axis='x', which='major', **{'gridOn': False, 'tick1On': True, 'tick2On': False, 'label1On': True, 'label2On': False, 'size': 6, 'tickdir': 'out', 'width': 1})
    axes.tick_params(axis='y', which='major', **{'gridOn': False, 'tick1On': True, 'tick2On': False, 'label1On': True, 'label2On': False, 'size': 6, 'tickdir': 'out', 'width': 1})
    axes.set_title(ws_name)
    axes.set_xlabel('Scattering angle ($degrees$)')
    axes.set_ylabel('Counts ($degrees$)$^{-1}$')
    axes.set_xlim([5.0, 12.0])
    axes.set_ylim([-5.0, max_range])
    axes.legend().draggable()

    plt.show()

# CORRECT BACKGROUND REMOVAL CODE:
Load(Filename='/archive/ndxlet/Instrument/data/cycle_21_1/LET00073023.nxs', OutputWorkspace='WB_LET073023_s', LoadMonitors='Separate')
Load(Filename='/archive/ndxlet/Instrument/data/cycle_21_1/LET00073020.nxs', OutputWorkspace='EBG4WB_LET073020_s', LoadMonitors='Separate')
CreateSingleValuedWorkspace(OutputWorkspace='__python_binary_op_single_value', DataValue=2.0856953905333242)
Multiply(LHSWorkspace='EBG4WB_LET073020_s', RHSWorkspace='__python_binary_op_single_value', OutputWorkspace='ebg_ws')
Minus(LHSWorkspace='WB_LET073023_s', RHSWorkspace='ebg_ws', OutputWorkspace='WB_LET073023_s')
#
ConvertUnits(InputWorkspace='WB_LET073023_s',OutputWorkspace='WB_LET073023_s',Target='Energy')
Rebin(InputWorkspace='WB_LET073023_s',OutputWorkspace='WB_LET073023_s',Params='4,100,20') # This is correctly bg-removed workspace
plot_ws('WB_LET073023_s',1,100000)
#<-------- Correct background removal ends here
BuildIncorrect = True

#  ACTUAL (simplified) CODE, produced from data reduction history, and producing incorrect result
Load(Filename='/archive/ndxlet/Instrument/data/cycle_21_1/LET00073024.nxs', OutputWorkspace='SR_LET073024', LoadMonitors=True, MonitorsLoadOnly='Histogram')
Load(Filename='/archive/ndxlet/Instrument/data/cycle_21_1/LET00073023.nxs', OutputWorkspace='WB_LET073023', LoadMonitors='Separate')
Load(Filename='/archive/ndxlet/Instrument/data/cycle_21_1/LET00073022.nxs', OutputWorkspace='EBG_LET073022', LoadMonitors=True, MonitorsLoadOnly='Histogram')

#- INTERFERING BLOCK 1--------------------------------------------------------------------------->
if BuildIncorrect:
    CreateSingleValuedWorkspace(OutputWorkspace='__python_binary_op_single_value', DataValue=5.9538491901525914)
    Multiply(LHSWorkspace='EBG_LET073022', RHSWorkspace='__python_binary_op_single_value', OutputWorkspace='ebg_ws')
    Minus(LHSWorkspace='SR_LET073024', RHSWorkspace='ebg_ws', OutputWorkspace='SR_LET073024')
#<--------------------------------------------------------------------------------------------------
Load(Filename='/archive/ndxlet/Instrument/data/cycle_21_1/LET00073020.nxs', OutputWorkspace='EBG4WB_LET073020', LoadMonitors='Separate')
#--INTERFERING BLOCK 2 --------------------------------------------------------------------------------------->
CreateSingleValuedWorkspace(OutputWorkspace='__python_binary_op_single_value', DataValue=2.0856953905333242)
Multiply(LHSWorkspace='EBG4WB_LET073020', RHSWorkspace='__python_binary_op_single_value', OutputWorkspace='ebg_ws')
Minus(LHSWorkspace='WB_LET073023', RHSWorkspace='ebg_ws', OutputWorkspace='WB_LET073023')
#<--------------------------------------------------------------------------------------------------

ConvertUnits(InputWorkspace='WB_LET073023', OutputWorkspace='WB_LET073023', Target='Energy')
Rebin(InputWorkspace='WB_LET073023', OutputWorkspace='WB_LET073023_normalized_cc', Params='4,32,20', IgnoreBinErrors=True)
plot_ws('WB_LET073023_normalized_cc',2,100000)
```
### Actual behavior
If fact, these workspaces are different:
[image]

If I copy the incorrect workspace (`WB_LET073023_normalized_cc`) creation history into script (below), I will produce the correct workspace:
[image]

The correct removal script below is actually equivalent to the one on top of the script above. In addition to that, if I modify the top script, setting `BuildIncorrect` to False, the top script is producing two correct images. The issue looks like entire mystery to me, but I can not trust my Python code anymore. 
```Python
# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np
from mantid.api import AnalysisDataService as ADS

def plot_ws(ws_name,fig_num,max_range):
    ConvertSpectrumAxis(InputWorkspace=ws_name,OutputWorkspace='_PWS',Target='Theta')
    Transpose(InputWorkspace='_PWS',OutputWorkspace=ws_name+'_PWS')

    pws = ADS.retrieve(ws_name+'_PWS')

    fig = plt.figure(fig_num)
    axes = fig.add_subplot(111, projection="mantid")
    #fig, axes = plt.subplots(num=1, subplot_kw={'projection': 'mantid'})
    axes.plot(pws, color='#1f77b4', label=ws_name, linestyle='None', marker='.', markersize=1.0, wkspIndex=0, zorder=2.1)
    axes.tick_params(axis='x', which='major', **{'gridOn': False, 'tick1On': True, 'tick2On': False, 'label1On': True, 'label2On': False, 'size': 6, 'tickdir': 'out', 'width': 1})
    axes.tick_params(axis='y', which='major', **{'gridOn': False, 'tick1On': True, 'tick2On': False, 'label1On': True, 'label2On': False, 'size': 6, 'tickdir': 'out', 'width': 1})
    axes.set_title(ws_name)
    axes.set_xlabel('Scattering angle ($degrees$)')
    axes.set_ylabel('Counts ($degrees$)$^{-1}$')
    axes.set_xlim([5.0, 12.0])
    axes.set_ylim([-5.0, max_range])
    axes.legend().draggable()

    plt.show()

Load(Filename='/archive/ndxlet/Instrument/data/cycle_21_1/LET00073023.nxs', OutputWorkspace='WB_LET073023', LoadMonitors='Separate')
Load(Filename='/archive/ndxlet/Instrument/data/cycle_21_1/LET00073020.nxs', OutputWorkspace='EBG4WB_LET073020', LoadMonitors='Separate')
CreateSingleValuedWorkspace(OutputWorkspace='__python_binary_op_single_value', DataValue=2.0856953905333242)
Multiply(LHSWorkspace='EBG4WB_LET073020', RHSWorkspace='__python_binary_op_single_value', OutputWorkspace='ebg_ws')
Minus(LHSWorkspace='WB_LET073023', RHSWorkspace='ebg_ws', OutputWorkspace='WB_LET073023')
ConvertUnits(InputWorkspace='WB_LET073023', OutputWorkspace='WB_LET073023', Target='Energy')
Rebin(InputWorkspace='WB_LET073023', OutputWorkspace='WB_LET073023_normalized_cc', Params='4,32,20', IgnoreBinErrors=True)

plot_ws('WB_LET073023_normalized_cc',3,100000)

```

### Steps to reproduce the behavior
Scripts provided. I was not able to reproduce this behavior using testing workspaces. Something is related to actual workspaces here. 

### Platforms affected
All

SimonHeybrock, 2021-06-18:
Could this be related to any of the issues documented in https://github.com/mantidproject/mantid/pull/14686?

martyngigg, 2021-06-18:
Possibly. Without a deeper look at the moment I can't say but it is most likely ADS interference I guess. I'll get someone to look at this next sprint.

martyngigg, 2021-06-18:
If I insert a `DeleteWorkspace("ebg_ws")` at the end of the `BuildIncorrect` block then both plots match so it's the presence of that workspace specifically that seems to interfere...

abuts, 2021-06-18:
yes, this is the fix I did in reduction script (#31806)

closed 2021-07-27 by martyngigg