mantidproject/mantid #31858: OS dependent grouping in SmoothNeighbours

RichardWaiteSTFC, 2021-06-22:
### Expected behavior
`SmoothNeighbours` produce the same output on all OS

### Actual behavior
In investigating #31857 it appears the grouping is different on mac and windows/RHEL.

The spectrum with workspace index 19397 in the WISH test data "WISH19612_vana_bank1_SXProcessed.nxs" generated on IDAaaS (RHEL) doesn't match the mac produced workspace in the test `ISIS_WISHSingleCrystalReduction.WISHProcessVanadiumForNormalisationTest` (in the image the mismatch workspace was from a mac jenkins build)

[image]

On instrument view it looks like it has changed the detector grouping
[image]

### Steps to reproduce the behavior
(1) On Mac run the systems test
systemtest.bat -C debug -R ISIS_WISHSingleCrystalReduction.WISHProcessVanadiumForNormalisationTest -j 2

Note is PR #31857 has been merged you will need to decrease the tolerance.

### Platforms affected

DannyHindson, 2021-06-29:
Part of the problem here is that due to the grouping of detectors onto spectra, any "edge" spectra have their closest neighbour spectra arranged such that the 2nd and 3rd are almost the same distance and same for the 8th\9th. This is a screenshot from the instrument viewer with some annotations added over the top by me. The rectangle labelled "centre" is spectrum 9 and the annotations show the ranking of the neighbouring spectra relative to spectrum 9:

[image]

So if you pick the nearest 8 neighbours you run the risk of floating point precision differences picking a different 8th neighbour. I think the number of nearest neighbours was increased to 8 recently by the SNS. It may be that picking the nearest 6 might be better for WISH.

RichardWaiteSTFC, 2021-06-29:
> So if you pick the nearest 8 neighbours you run the risk of floating point precision differences picking a different 8th neighbour. I think the number of nearest neighbours was increased to 8 recently by the SNS. It may be that picking the nearest 6 might be better for WISH.

I seem to remember the number of nearest neighbours being 5 before (in DetectorSearch?) - which from looking at your nice diagram would still work, but perhaps 6 is better. I don't suppose it matters?

DannyHindson, 2021-06-30:
Actually I got my wires crossed a bit there. The change the SNS did was to a different class - DetectorSearch. Although it sounds similar, that's not used by SmoothNeighbours. The number of neighbours used in SmoothNeighbours is an input parameter to the algorithm that has a default of 8 that's always been there. I guess that makes sense for a regular rectangular array.
So I think I'm going to fix this by just setting number of neighbours to 6. I expect I'll need to update the reference file in this test as a result. The floating point rounding I was doing had an arbitrary parameter of the number of decimal places to round to so that was a bit nasty.

RichardWaiteSTFC, 2021-06-30:
> So I think I'm going to fix this by just setting number of neighbours to 6. I expect I'll need to update the reference file in this test as a result. The floating point rounding I was doing had an arbitrary parameter of the number of decimal places to round to so that was a bit nasty.

Sounds good. As long as the grouping on a given OS is the same every time I don't think WISH will need to update their script right? But maybe we should warn them that there is some OS specific behaviour.

DannyHindson, 2021-06-30:
I'll let them know about this change. It's not essential they change if they use the same OS all the time but it's nice I guess if our system test corresponds to their script so I'll suggest they probably should update.

DannyHindson, 2021-06-30:
Just for my notes in case I come back to this. I wondered whether the different floating point results on different platforms might be resolved by using of a compiler option. On clang (which I think is used on Mac) there is an option called -ffp-contract that determines whether the compiler uses more precision than "double" for intermediate results and then reduces the result to the required precision at the last step. There is a similar option on Windows that you can set via "#pragma fp_contract (off)" and I assume a similar feature on GCC. If I get chance I may check to see if this aligns the fp behaviour on the different platforms

PR #31932 (2021-07-05): Fix ISIS_WISHSingleCrystalReduction system test
**Description of work.**

A temporary fix was put into the ISIS_WISHSingleCrystalReduction.WISHProcessVanadiumForNormalisationTest in https://github.com/mantidproject/mantid/pull/31857 to ensure the result was the same on all platforms (the Mac result was different).

This PR implements a permanent fix by change the value of the NumberOfNeighbours parameter to 6 instead of the default value of 8. The centres of the spectra on WISH are arranged in more of a hexagonal way than a standard rectangular grid. This means that the 8th and 9th nearest nei [... cut]
closed 2021-07-05 by gvardany