mantidproject/mantid #31863: Workspace calculator throws errors when clearing ADS

RichardWaiteSTFC, 2021-06-23:
### Expected behavior
Clearing the ADS (or otherwise deleting a workspace would not cause workspace calculator to throw an error). 

### Actual behavior
It looks like the calculator populates itself automatically with a workspace and when clearing the ADS it throws an error (even when the widget is not visible - it looks like it's being run on clear).
```
The LHS input workspace does not exist.
The RHS input workspace does not exist.
The LHS input workspace does not exist.
The RHS input workspace does not exist.
```
If you only delete the workspace in the calculator it doesn't throw an error - but neither does it remove the workspace form the calculator so clicking Go will throw an error.

### Steps to reproduce the behavior
(1) Make a workspace
`ws = CreateSampleWorkspace(Function="User Defined",   UserDefinedFunction="name=LinearBackground, A0=0.3;name=Gaussian, PeakCentre=5, Height=10, Sigma=0.7",  NumBanks=1, BankPixelWidth=1, XMin=0, XMax=10,  BinWidth=0.1, WorkspaceType="Event")`
(2) Clear ADS

### Platforms affected

DannyHindson, 2021-06-23:
@darominski - please can you take a look at this? It seems like the new widget generates errors in the log if the user clears all the workspaces, even if they've hidden the workspace calculator widget. I think the widget should probably quietly clear the lhs or rhs dropdown without creating an error if the dropdowns contain a workspace that gets deleted.

darominski, 2021-06-23:
Yes, absolutely. I am not sure why does the `WorkspaceSelector` moves from its default state (empty) to selecting one of the workspaces in the ADS. In any case, I propose to supress the error logs altogether. The related PR id is: #31867 .

PR #31867 (2021-06-23): Silence validation for WorkspaceCalculatorWidget
It has been noticed that the error logs from the `WorkspaceCalculatorWidget` are showing up on ADS clear event, even if no interaction with the widget has taken place. There is a suspected automatic selection of a a workspace present in the ADS in one of `WorkspaceSelector`'s, which on a clear event fails validation, as it does not exist, and yields an error message. 

The proposed solution is removal of the error logs altogether from the `WorkspaceSelectorWidget` and depending on the commucation of issues through asterisk tooltip located next t [... cut]
closed 2021-06-23 by martyngigg