mantidproject/mantid #31876: Muon Analysis - Extend Fitted line beyond data points option

robertapplin, 2021-06-24:
### Description
The ability to extend the fitted line on a plot beyond the data points has been requested by Muon scientists. This would be useful in model fitting where there are a small number of data points, and they want to see the trend of the line beyond these data points. This is **not** a fitting feature, but rather a plotting feature. 

To do this, the fit function at the end of a fit will need to be stored in the `FitInformation` and then passed to the plotting model. The plotting code will then Evaluate the fit function between a certain x range (maybe the x range of the plot), and using a given number of steps in the x value. This can been done using the `EvaluateFunction` algorithm.

So to sum up we need two new plotting options:
- A checkbox to turn on extending the fit line to the x limits of the plot
- A line edit to allow users to enter the number of x steps to evaluate for the extended fitted line. This should be disabled if the checkbox is not ticked.

AnthonyLim23, 2021-07-26:
This is on hold

AnthonyLim23, 2021-09-24:
Will need to talk to Francis and Steve about the design. It might be worth replacing the `plot guess` with a new:

`<tick box> plot function with <line edit> data points`

and take the range from the start and end x on the fitting pane. I suggest a mock up of this to ask if its ok

PR #33108 (2021-12-16): Extend guess range outside of data range in Muon analysis
**Description of work.**

This PR adds the ability to plot the fit guess outside of the range of the data in muon analysis.

**To test:**
1. Open muon analysis.
2. load MUSR 62260-3
3. go to fitting
4. add gausosc function
5. check plot guess by default it plots using the same range as the data
6. in the fit options change the `plot guess using` to x at data points. this allows you to chane the number of points to be used in the guess workspace
7. chane the number of points and the plot will update.
8. change the `plot guess using` to c [... cut]
closed 2021-12-16 by DavidFair