mantidproject/mantid #31889: QENS data analysis interface

SilkeSchomann, 2021-06-25:
This issue is connected to #[31508](https://github.com/mantidproject/mantid/issues/31508)

- [ ] Rename 'Inelastic Data Analysis' to 'QENS Data Analysis'
- [ ] Ensure help page is available
- [ ] Mention on help page how to remove the restriction to *red.nxs and *sqw.nxs for file names through the settings
- [ ]  If QENS data analysis remains available both in the direct and the indirect interface, it would be good if 'Restrict allowed input files by name (recommended)' in the settings is by default enabled for the indirect interface and disabled for the direct interface. This is optional and would be obsolete if #31508 is done. It is worthwhile checking which option is the best and whether this is possible.

SpencerHowells, 2021-06-25:
@SilkeSchomann Item 1. This is incorrect it is not restricted to QENS!

SpencerHowells, 2021-06-28:
@SilkeSchomann Item 3:  on web page https://docs.mantidproject.org/nightly/interfaces/indirect/Inelastic%20Data%20Analysis.html
it says 
**Settings**
Opens the Settings GUI which allows you to customize the settings for the Indirect interfaces.
Clicking on Settings (green) takes you to 
https://docs.mantidproject.org/nightly/interfaces/indirect/Indirect%20Settings.html#interface-indirect-settings
which gives a full description

closed 2022-07-07 by stale[bot]