mantidproject/mantid #31898: Muon easy fitting

AnthonyLim23, 2021-06-28:
If the user is fitting multiple functions they may know that the asymmetries (amplitudes) must add up to a specific value. In Mantid this is called a tie. However, users would like it to be easier to create these dependencies. 

Initial = relaxing + baseline

The sum of the amplitudes of the components is the relaxing value

The initial asymmetry is the total asymmetry at t=0.
The baseline is related to the background

We have:

f(t) = sum_j  A_j* G_j(t)*R_j(t) + B*exp[(- lambda*t)^beta]

Such that:

Initial = B + sum_j A_j

As a script this looks like:

```
# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

#set up
x = np.linspace(0,10,500)

A0 = .25
A1 = .12
A2 = .5
total = A0+A1+A2

fun = (GausOsc(Frequency = 1.3, A=A0) + GausOsc(Frequency = 1.9, A=A1) + FlatBackground(A0=A2))
y = fun(x)*(1.+.2*np.random.normal(0, 0.8, len(x)))
ws = CreateWorkspace(x,y)

relaxing_sum = '(f0.A + f1.A)'
BG = 'f2.A0'
I0 = str(total)
relax_val = str(A0+A1)

# create fit func
test = GausOsc(Frequency = 1.3) + GausOsc(Frequency = 1.) + FlatBackground()

# fix the total asymmetry to a constant
test.tie({BG : I0 + '-'+ relaxing_sum})

# fix BG
#test.tie({BG: A2})

# fix relaxing (sum of Asym's)
#test.tie({'f0.A' : relax_val +'- f1.A'})

out = Fit(test,ws, OutputWorkspace="test", createOutput=True)

table = out.OutputParameters

i  =1
print( table.column(i)[0], table.column(i)[4], table.column(i)[8])
print( table.column(i)[0]+ table.column(i)[4]+ table.column(i)[8], total)
```

AnthonyLim23, 2021-06-28:
potential mock up (this should be used as a "add Function" alternative and will output the fit function into the function browser

[image]

AnthonyLim23, 2021-06-28:
we will need a list of functions to include in this interface and which parameters will need to be editable.

robertapplin, 2021-06-28:
I would say it would take around 20-25 working days to do this, which includes time taken to create a new GUI (fully tested) and integrate it into the existing setup. Also it seems like a piece of work that may have edge cases and slight nuisances that will need ironing out or including

closed 2024-05-21 by github-actions[bot]