mantidproject/mantid #31943: Monochromatic SANS at ILL

gvardany, 2021-07-01:
The current SANS workflow is built such that it works both for TOF and monochromatic modes.
While it is optimal for TOF, it is not optimal for monochromatic, since the x-axis dimension is not utilized.
One could use that dimension to store time frames for kinetic monochromatic, sample environment parameter scans and multiple different samples in case of standard SANS.
This will greatly reduce the number of workspaces and groups that are created/deleted in the process.
This will also eliminate the redundancy of the in-memory instrument object being copied for each sample and result in vast memory usage reduction, which will in turn speed up the runtime due to avoiding allocation and deallocation of the memory for the instrument clones.
The downside of this (and the original reason why it wasn't done in the first place), is that certain child algorithms required in the workflow need to be reworked to accept also 2D workspaces, where the x-unit is not the wavelength, but is just an integer frame index. This will allow outsourcing, for example, the loop over samples to the C++ child algorithms, rather than in the master python workflow, which is what is done currently and proves to be slow. Some child algorithms will need to process the input 2D workspace frame-wise, i.e. bin-by-bin, and produce individual outputs by stacking them along the spectrum axis of the output workspace. The spectrum axis can then be replaces by a text axis to reflect the names of the samples.

 The master plan for this is as follows:

- [ ] Add choice CombinationType=[Merge (default), Concatenate] to LoadAndMerge

- [ ] Refactor/extend the following algorithms to take 2D, but monochromatic inputs:
      1. CalculateTransmission
      2. ApplyTransmissionCorrection
      3. Q1DWeighetd (this should also be renamed to AzimuthalIntegration)
      4. Qxy
      ... 

- [ ] Devise a new algorithm MonoSANSILLMultiProcess that will take multiple samples at multiple distances (potentially also summed), put all samples (kinetic will come out like that already from the loader) in singe 2D workspaces (per distance), correct those workspaces for all the instrumental effects and then pass to the integration.

This also lays the groundwork for the advanced sans (i.e. sample scans), where the corrections will be done with this optimal algorithm, then the resulting semi-reduced workspace can be graphically integrated in a planned new tab of the instrument viewer - SANS showing the I(Q) (potentially per selected shape) in the instrument viewer.

martyngigg, 2022-01-10:
I was just moving uncompleted feature issues over to 6.4 but I notice this one has all tasks marked as done so is it ready to be closed?

gvardany, 2022-01-10:
Nope, it's ready to be reviewed :)

PR #32287 (2022-02-15): ILL SANS v2
**Description of work.**
This is the refactoring of ILL SANS suite resulting in 1-2 orders of magnitude of speed-up for monochromatic/kinetic.
The principal change is the concatenation of multiple runs/frames along the x-axis into a single workspace (per detector distance).
Introduces a v2 of `SANSILLReduction`, which supersedes and deprecates the v1.
Introduces a new algorithm `SANSILLMultiProcess`, which deprecates `SANSILLAutoProcess`.
The 2 are tightly coupled, so they had to be written together.
The clean-up of the loader, however, will be done in a separate PR.

**For reviewe [... cut]
closed 2022-02-15 by martyngigg